sccn/eeglab #141: Spectra and Maps error

ksk-2, 2020-03-06:
### Description
Not able to generate Spectra and Maps through the plot options. The data is a muse csv file. 
<br>
#### Steps to Reproduce

1. Import muse data
2. Set locations
3. Run ICA
4. Under Plot select channel or component maps
5. confirm options

#### Expected behavior:
Spectral maps with freq and power info along with 2D images

<br>
#### Actual behavior:
Error. See attached image. 

<br>
#### Versions
latest version

|   |   |
|----| ---- | 
| OS version     | [windows 10] |
| Matlab version | [2019] |
| EEGLAB version | [2019] |

[image]

ksk-2, 2020-03-10:
Can anyone shed some light on this? Thanks!

arnodelorme, 2020-04-07:
This is because you do not have the Matlab signal processing toolbox installed. However, the function should use alternative methods. What does this command return on your machine?

usepwelch = license('checkout','Signal_Toolbox')

arnodelorme, 2020-04-07:
And apologies for replying so late. You replied to your own message which led us to believe that your concern had been addressed by someone from our team.

cbpickro13, 2020-04-24:
Hi, 
I've run into the same issue EEGLab error in function spectcomp() at line 941. 
Here is what it says in the Maltab command window... 
C:\Program Files\MATLAB\R2020a\toolbox\stats\stats\finv.m
Pop_spectopo: finding data discontinuities
Selecting the first 15.0% of data for analysis...

I tried running usepwelch = license('checkout','Signal_Toolbox') and I received usepwlch = 1 as the return.

qwezy923, 2021-04-13:
Hello, I also got stuck in this issue, and tried your method from #119, but I cannot find the path to signal processing toolbox.
After running usepwelch = license('checkout','Signal_Toolbox') , I still got usepwlch = 1.
Thanks.

arnodelorme, 2022-04-19:
This is because it is available on your network but you cannot check it out (someone else is using it).

closed 2022-04-19 by arnodelorme