sccn/eeglab #147: Error in function std_ersp

Naama2020, 2020-03-27:
Hi,
I am trying to pre-compute component measures in my study of 31 subjects and I get the attached error message when I mark ERSP option. Does anyone know what might be leading to this problem and what the solution is?

[*Description of the bug or feature
Attach a link to Dropbox or Google Drive with screenshots and your data if possible.*]

<br>
#### Steps to Reproduce

1. Uploaded 31 datasets - 1 dataset per subject
2. Created the study and chose the 'select by r.v.' option (and hit ok for 15%)
3. Defined the study design (2*2 within sub.)
4. In precompute component measures window selected ERP, Spectra, ERSP, ITC, and scalp map options.

<br>
#### Expected behavior: 
[*What you expected to happen*]

<br>
#### Actual behavior: the function is stopped with attached function error
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [Windows 10 64 bit] |
| Matlab version | [2019b] |
| EEGLAB version | [2019.1 dev
[image]

] |

arnodelorme, 2020-03-27:
Fixed the issue with e35d7ab80. Due to a missing toolbox. Let us know if it works better. You can also set 'parallel' to 'off' to avoid the issue.

closed 2020-03-27 by arnodelorme

Naama2020, 2020-03-29:
I did as instructed. I tried inserting the indicated code into the function and recieved the same error message (parallel is set to 'off').

arnodelorme, 2020-03-29:
Did you try with the corrected function (attached)?

Best wishes,

Arno

Naama2020, 2020-03-29:
This is what the indicated section of the function looks like now:

% CHANGE THE LINE BELOW TO PARFOR TO USE THE PARALLEL TOOLBOX
disp('Computing time/frequency decomposition...');
numWorkers = 0;
try
    if strcmpi(g.parallel, 'off') && ~exist('parpool') && exist('gcp')
        poolobj = gcp('nocreate');
        if ~isempty(poolobj)
            numWorkers = poolobj.NumWorkers;
        end
    elseif strcmpi(g.parallel, 'on')
        try
            myCluster = parcluster('local');
            numWorkers = myCluster.NumWorkers;
        catch, disp('Cound not start parallel job'); end
        end
    catch
    g.parallel = 'off';
    
end

arnodelorme, 2020-03-29:
Yes, the clause 

poolobj = gcp('nocreate’);

Is in a try, catch sequence, so it cannot generate the error as before. You must be using another version of the function.
Could you type on the command line

which(’std_ersp’)

Best wishes,

Arno

Naama2020, 2020-03-29:
Is this what you mean?

which(std_ersp)
  std_ersp() - Compute ERSP and/or ITC transforms for ICA components 
               or data channels of a dataset. Save results into Matlab 
               float files. 
 
  Function description:
               The function computes the mean ERSP or ITC for the selected 
               dataset ICA components or data channels in the requested 
               frequency range and time window (the two are dependent). 
               Frequencies are equally log spaced. Options specify component 
               numbers, desired frequency range,  time window length, 
               frequency resolution, significance level, and wavelet
               cycles. See >> help newtimef and >> timef details 
 
               Two Matlab files are saved (for ERSP and ITC). These contain 
               the ERSP|ITC image, plus the transform parameters 
               used to compute them. Saves the computed dataset mean images 
               in dataset-name files with extensions '.icaersp' and '.icaitc'
               for ICA components or '.datersp', '.datitc' for data channels.
  Usage:  
               >> [X times logfreqs ] = std_ersp(EEG, 'key', 'val', ...);
  Inputs:
    EEG          - a loaded epoched EEG dataset structure. May be an array
                   of such structure containing several datasets.
 
  Other inputs:
    'trialindices' - [cell array] indices of trials for each dataset.
                   Default is EMPTY (no trials). NEEDS TO BE SET.
    'components' - [numeric vector] components of the EEG structure for which 
                   activation spectrum will be computed. Note that because 
                   computation of ERP is so fast, all components spectrum are
                   computed and saved. Only selected component 
                   are returned by the function to Matlab
                   {default|[] -> all}
    'channels'   - [cell array] channels of the EEG structure for which 
                   activation spectrum will be computed. Note that because 
                   computation of ERP is so fast, all channels spectrum are
                   computed and saved. Only selected channels 
                   are returned by the function to Matlab
                   {default|[] -> none}
    'recompute'  - ['on'|'off'] force recomputing ERP file even if it is 
                   already on disk.
    'recompute'  - ['on'|'off'] force recomputing data file even if it is 
                   already on disk.
    'rmcomps'    - [integer array] remove artifactual components (this entry
                   is ignored when plotting components). This entry contains 
                   the indices of the components to be removed. Default is none.
    'interp'     - [struct] channel location structure containing electrode
                   to interpolate ((this entry is ignored when plotting 
                   components). Default is no interpolation.
    'fileout'    - [string] name of the file to save on disk. The default
                   is the same name (with a different extension) as the 
                   dataset given as input.
   'savetrials'  - ['on'|'off'] save single-trials ERSP. Requires a lot of disk
                   space (dataset space on disk times 10) but allow for refined
                   single-trial statistics. This option is obsolete as
                   trials are now always saved.
   'savefile'    - ['on'|'off'] save file or simply return measures.
                   Default is to save files ('on').
   'getparams'   - ['on'|'off'] return optional parameters for the newtimef 
                   function (and do not compute anything). This argument is
                   obsolete (default is 'off').
 
  ERSP optional inputs:
    'type'       - ['ersp'|'itc'|'ersp&itc'] save ERSP, ITC, or both data 
                   types to disk {default: 'ersp'}
    'freqs'      - [minHz maxHz] the ERSP/ITC frequency range to compute 
                   and return. {default: 3 to EEG sampling rate divided by 3}
[... 29 lines cut ...]
    logfreqs  - vector of (equally log spaced) frequencies (in Hz) at which the 
                log ERSP/ITC was evaluated. 
    parameters - parameters given as input to the newtimef function.
 
  Files written or modified:     
               [dataset_filename].icaersp   <-- saved component ERSPs
               [dataset_filename].icaitc    <-- saved component ITCs
               [dataset_filename].icatimef  <-- saved component single
                                                trial decompositions.
   OR for channels
               [dataset_filename].datersp   <-- saved channel ERSPs
               [dataset_filename].datitc    <-- saved channel ITCs
               [dataset_filename].dattimef  <-- saved channel single
                                                trial decompositions.
  Example: 
             % Create mean ERSP and ITC images on disk for all comps from 
             % dataset EEG use three-cycle wavelets (at 3 Hz) to more than 
             % three-cycle wavelets at 50 Hz. See >> help newtimef 
             % Return the (equally log-freq spaced, probability-masked) ERSP.
             >> [Xersp, times, logfreqs] = std_ersp(EEG, ...
                        'type', 'ersp', 'freqs', [3 50], 'cycles', [3 0.5]);
 
  See also: timef(), std_itc(), std_erp(), std_spec(), std_topo(), std_preclust()
 
  Authors: Arnaud Delorme, Hilit Serby, SCCN, INC, UCSD, January, 2005-

arnodelorme, 2020-03-29:
You need to type with the quotes as in

 which('std_ersp')
/data/matlab/eeglab/functions/studyfunc/std_ersp.m

Best wishes,

Arno

> On Mar 28, 2020, at 9:10 PM, Naama2020 <notifications@github.com> wrote:
> 
> which(std_ersp)

Naama2020, 2020-03-29:
My bad. Looks like it is working. My mistake was that the changes I had made to the function were not saved earlier - but now I saved and looks like the function is running smoothly. No error message (however, it does take the function quite a while to run).

arnodelorme, 2020-03-29:
Great.
Best wishes,

Arno

Naba1210, 2022-09-09:
While Computing ERSP in a study, the errors include
[image]
[image]

[image]