sccn/eeglab #148: fread error message

Naama2020, 2020-03-27:
### Description

[*Hi,
I am trying to pre-compute component measures in my study of 31 subjects and I get the attached error message when I mark ERSP option. Does anyone know what might be leading to this problem and what the solution is?*]

<br>
#### Steps to Reproduce

1. Uploaded 31 datasets - 1 dataset per subject
2. Created the study and chose the 'select by r.v.' option (and hit ok for 15%)
3. Defined the study design (2*2 within sub.)
4. In precompute component measures window selected ERP, Spectra, and scalp map options.
5. In build preclustering array window selected ERP, Spectra, ERSP, ITC, and scalp map options.

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior: the function is stopped with attached function error.
[image]

[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [Windows 10 64 bit] |
| Matlab version | [2019b] |
| EEGLAB version | [2019.1 dev
[image]
] |

arnodelorme, 2020-04-01:
Would you mind to share your study. One possibility is that some files were not copied. For example EEGLAB datasets are made of 2 files (.set and .fdt files). If you only copy the .set file, this is the kind of error you would see.

closed 2020-04-01 by arnodelorme

Naama2020, 2020-04-04:
I shared my study, but it will take some time for the files to upload. The
link is attached. It does not seem to be because of the .fdt files -
they're all there.

https://drive.google.com/drive/folders/1yu8YO2OpuMeeFp9dWwv5K0GB-euhgb8a?usp=sharing

‫בתאריך יום ד׳, 1 באפר׳ 2020 ב-19:10 מאת ‪Arnaud Delorme‬‏ <‪
notifications@github.com‬‏>:‬

Naama2020, 2020-04-04:
Additionally, the error does not appear if I add ERSP and ITC in the pre-compute stage (see error #147).