sccn/eeglab #153: Cleanline EEGLab error in function

cbpickro13, 2020-04-24:
### Description
I tried running the plugin Cleanline on continuous data that has been filtered with a 1hz highpass filter. I received the following error: 
EEGLab error in funtion dpsschk() at line 28:
Undefined funtion 'dpss' for input arguments of type 'double'.

<br>
#### Steps to Reproduce

1. imported data using EGI plugin 
2. downsampled data to 256hz 
3. highpass filter 1hz
4. imported data location with EGI .spf file, set reference channel Cz
5.  clicked on Cleanline plugin left all settings to default (did try changing the window length and step size both = 4 but this resulted in same error).

I've tried looking around online and found a mention of a bug from 2014 that mentions the fieldtrip plugin, but I do not have this installed. It also mentions moving the matlab functions to higher up on the path list, however I don't know what that means, in my "current folder" window the subfolder functions is the first folder listed. 

|   |   |
|----| ---- | 
| OS version     | [PC] |
| Matlab version | [R2020a] |
| EEGLAB version | [2019.1] |

arnodelorme, 2020-05-18:
Cleanline only works with the Matlab signal processing toolbox (an additional paying toolbox) and apparently you are missing that function. I have added a note to the cleanline repository.

closed 2020-05-18 by arnodelorme

demetyesilbass, 2023-11-16:
> e repository.

Hi,

I also want to use cleanline and I also get an error that 

> Undefined function or variable 'lgamma'.

Now i am also live same problem and I have searched and I couldn't find a function called **_lgamma_**. Currently, I am using the last version of eeglab and I just wonder is that possible the function called **_lgamma_**  actually must be _**gammaIn**_?

Thanks and looking forward to hearing you.

Best,
Demet

demetyesilbass, 2023-11-16:
[image]