sccn/eeglab #156: no spectrum plotted

cereso69, 2020-05-07:
### Description

no spectrum plotted when inspect/label components by map

<br>
#### Steps to Reproduce

1. After Decompose data by ICA
2. Inspect/label components by map
3. Click on a component
4. See image for error - no spectrum plotted

<br>
#### Expected behavior:
spectrum plotted

<br>
#### Actual behavior:
no spectrum plotted

<br>
v2019.1

|   |   |
|----| ---- | 
| OS version     | [latest] |
| Matlab version | [2019b
[image]
] |
| EEGLAB version | [2019.1] |

nucleuscub, 2020-05-07:
It may be that you do not have the signal processing toolbox available in your MATLAB installation.
By typing in the MATLAB command window the following command, you should get the location of the pwelch function :  _which pwelch_ 
The output should resemble this: /Applications/MATLAB_R2018b.app/toolbox/signal/signal/pwelch.m

If not like this, you probably don't have the toolbox.
It should work without the toolbox, though. So we'll take a look.

nucleuscub, 2020-05-07:
@cereso69 , 
 Can you please paste the output of the following commands in the MATLAB command window?

license('checkout','Signal_Toolbox')
which pwelch

Thanks, 
R

cereso69, 2020-05-07:
>> license('checkout','Signal_Toolbox')
which pwelch

ans =

     1

'pwelch' not found.
>>

cereso69, 2020-05-07:
Thank you for helping!

nucleuscub, 2020-05-07:
Dear @cereso69 , 
 It seems you do have the Signal Processing Toolbox, but the pwelch function is missing (should be in this toolbox). I'll try reinstalling the Signal Processing Toolbox. 
Hope this helps, 
R

closed 2020-05-07 by nucleuscub

cereso69, 2020-05-07:
Hi R,

Thank you, but how do I re-install the signal processing toolbox please? Simple instructions would be much appreciated.

Thanks again