sccn/eeglab #159: plugin_getweb fails

tiborauer, 2020-05-13:
### Description

`plugin_getweb` fails upon start because of the unusual description of `detect_spindles`

<br>
#### Steps to Reproduce

1. `eeglab`

<br>
#### Expected behavior:
Some menus items hidden. Use Preference menu to show them all.
eeglab: options file is ~/eeg_options.m
EEGLAB warning: there can be only one EEGLAB window, closing old one
Retrieving plugin versions from server...
Retreiving download statistics...
...
 

<br>
#### Actual behavior:
Some menus items hidden. Use Preference menu to show them all.
eeglab: options file is ~/eeg_options.m
EEGLAB warning: there can be only one EEGLAB window, closing old one
Retrieving plugin versions from server...
Retreiving download statistics...
Index exceeds the number of array elements (32).

Error in plugin_getweb (line 62)
    plugin(iRow).downloads = stats{2}(iRow);

Error in eeglab (line 915)
        pluginTmp = plugin_getweb('', pluginlist, 'newlist');

Error in eeglabClass/load (line 39)
            eeglab;
 

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     |CentOS Linux release 7.4.1708 (Core)
| Matlab version | R2019b
| EEGLAB version | v2019.1 (github)

amydesroches, 2020-05-13:
Same issue here with Matlab R2020a, EEGLAB v 2019.1.

arnodelorme, 2020-05-13:
Yes, we have solved the issue which was with a corrupted online plugin. We will make the code more robust. Thank you for reporting the problem.

Best wishes,

Arno

closed 2020-05-13 by arnodelorme

yunshufan, 2021-09-23:
I encountered a similar issue today:
[image]
It wasn't there yesterday.
I'm using Matlab2021a and eeglab2019_1
Do I need to download and install the newest version of eeglab?

b000z, 2021-09-23:
[image]

I encountered almost this problem yesterday with matlab r2020a and eeglab2019_1. I followed the instructions both https://github.com/sccn/eeglab/issues/20 and https://www.mathworks.com/matlabcentral/answers/130359-how-do-i-change-the-java-virtual-machine-jvm-that-matlab-is-using-on-windows which led me astray and caused matlab not to open. I found this and realized I may need to simply update eeglab. Just got v2021_1 and the issue is solved.

I cannot understand why this problem occurred. Have been using the current versions without any problem for more than 1 year, and all of a sudden, yesterday this error presented.