sccn/eeglab #164: load existing data set

dkjones-eeg, 2020-05-19:
### Description

Trying to load an existing dataset using the EEGLABv2019.1 window, File>Load existing dataset, I get the error: wrong number of arguments. Project must be given as argument,
(Error occurred in function settings_OpeningFcn() in line 69)

Screen shot is at
https://drive.google.com/file/d/1ZPJ3iQNBlilt5eAIUGeQkHx-wbgbbaS8/view?usp=sharing
<br>

#### Steps to Reproduce

1. launch eeglab
2. at the main eeglab window, in the file pull-down menu, select Load existing dataset
3. the error occurs as explained above

[image]

<br>
#### Expected behavior:
the dataset should load

<br>
#### Actual behavior:
this error occurs, and no dataset is loaded

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | Windows 10 |
| Matlab version | R2020a - academic use |
| EEGLAB version | eeglab2019_1 |

arnodelorme, 2020-05-22:
Never seen that error. Even after digging, could not find anything. The most likely problem is that there is another toolbox you have installed which has a GUI function called "settings.m" that corrupts the environment.

closed 2020-05-22 by arnodelorme

dkjones-eeg, 2020-05-26:
OK - this is a very useful hint. I'll check it out. Thanks!

Best Regards,
Daniel K. Jones, Ph.D., P.E.
Associate Professor, Mechanical Engineering Technology
SUNY Polytechnic Institute
100 Seymour Road
Utica, NY 13502
(315)792-7121 or 7234
Office hours and meetings available by appointment
1207 Donovan Hall

dkjones-eeg, 2020-05-28:
OK - I uninstalled matlab R2019 and R2020, along with all toolboxes. Then
reinstalled matlab R2020 with no toolboxes. The same error occurs. Please
let me know if you have any suggestions.

Best Regards,
Daniel K. Jones, Ph.D., P.E.
Associate Professor, Mechanical Engineering Technology
SUNY Polytechnic Institute
100 Seymour Road
Utica, NY 13502
(315)792-7121 or 7234
Office hours and meetings available by appointment
1207 Donovan Hall