sccn/eeglab #168: EDF+ Annotation timestamps missing submillisecond accuracy

greydongilmore, 2020-06-08:
It was noticed that the onset time for annotations in an EDF+ file might be missing the submillisecond accuracy associated with the start time of the recording. This results in annotation timestamps being delayed from the true event time. 

This annotation timestamp issue has been raised with [MNE-python](https://github.com/mne-tools/mne-python/issues/7872#issue-632648353) and [FieldTrip](https://github.com/fieldtrip/fieldtrip/issues/1425#issue-634921091) as well.

I will provide an explanation from both bug reports for completeness.

#### Expected results

This is the [EDF+ file](https://drive.google.com/file/d/1yql0bMGcXXrEZr1At_Cp-eMi6rTKSr_F/view?usp=sharing) used to produce this behavior.

If the EDF+ file is brought into [EDFbrowser](https://www.teuniz.net/edfbrowser/) it will be noticed that the millisecond aspect of the start time is displayed in the `Start` field:

[image]

For instance, this results in the annotation timestamps being:

[image]

#### Actual results

When loading the data into EEGlab the annotation timestamps return:

[image][image]

If you take the difference between the EEGlab timestamp and the EDFbrowser timestamp it is equal to the submillisecond portion of the recording start time.

#### Additional information

I reached out to Teunis van Beelen, the developer of [EDFbrowser](https://www.teuniz.net/edfbrowser/), to get an explanation. The explanation provided was as follows:

> EDF does not have support for subsecond precision of the starttime but EDF+ does. Because EDF+ needed to be backwards compatible with EDF and EDF doesn't have the possibility to write subsecond precision for the starttime into the header, the subsecond part of the starttime is written outside of the EDF header. The way the subsecond part of the starttime is stored in EDF+ is described here: https://www.edfplus.info/specs/edfplus.html#timekeeping

Teunis van Beelen provided a quote from the EDF specs:

> "The startdate/time of a file is specified in the EDF+ header fields 'startdate of recording' and 'starttime of recording'. These fields must indicate the absolute second in which the start of the first data record falls. So, the first TAL in the first data record always starts with +0.X2020, indicating that the first data record starts a fraction, X, of a second after the startdate/time that is specified in the EDF+ header. If X=0, then the .X may be omitted."

In other words, the subsecond part of the starttime is written in the EDF Annotations "signal".

#### Versions

* OS version: Ubuntu 20.04 LTS
* Matlab version: 2019b
* EEGLAB version: current master branch

arnodelorme, 2020-06-16:
Dear Grey, yes I understand the issue. Let's see what @schloegl has to say about it.

schloegl, 2020-06-16:
I'm aware of this issue. EDF+ is using really an awful hack here. GDF does not have that problem and provides 20us accuracy. 

In the repository of Biosig/biosig4c++ this issue is already addressed, it will be part of the next release (I'm working on the release). In order to address this within EEGlab, you need to build mexSOPEN/mexSLOAD and included in the path, so that sload.m can make use of these mex-functions.

greydongilmore, 2020-06-16:
The EDF developers provided this hack to make EDF+ backward compatible to EDF. Agreed it's not the best solution but it keeps EDF/EDF+ headers the same.

amisepa, 2021-03-01:
Hi @greydongilmore, do you still have this issue? 

If so, the EEGLAB EDF plugin has been recently updated and provides the option to use the mexSLOAD solution mentioned above. Just tick the 'Import using alternative BIOSIG method (mexSLOAD)'. 

Let me know if that works.

cll008, 2021-05-24:
Thank you all for reporting and solving the issue. I'll close this for now but please reopen if there are further comments.

closed 2021-05-24 by cll008