sccn/eeglab #170: Events import bug

mzleena, 2020-06-11:
When I try to upload an events file I get error: line 1149 eegplot error.
How to solve this problem.?

[*Description of the bug or feature
Attach a link to Dropbox or Google Drive with screenshots and your data if possible.*]

<br>
#### Steps to Reproduce

1. Import data file 
[Uploading eeglabsubj2.txt…]

2. Import Event file 
[Events_sub2_subtract.txt](https://github.com/sccn/eeglab/files/4767757/Events_sub2_subtract.txt)

3. Go to Plot menu, channel data scroll

Error Message shows up 
EEGLab error in function eegplot() line 1149 
Matrix dimensions must agree

<br>
#### Expected behavior:
I expect it to plot the data with the event markers in it.

<br>
#### Actual behavior:
Error message (see above)

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | 2019b |
| EEGLAB version | 2019.1 |

arnodelorme, 2020-06-16:
I have tried again with no problem, creating a dataset with 100,000 data points.

data = rand(4, 100000);
EEG = pop_importdata('dataformat','array','nbchan',0,'data','data','srate',1000,'pnts',0,'xmin',0);
EEG = pop_importevent( EEG, 'event','/Users/arno/Downloads/Events_sub2_subtract.txt','fields',{'latency','type','duration'},'timeunit',NaN);

arnodelorme, 2020-06-16:
[image]

closed 2020-06-16 by arnodelorme