sccn/eeglab #171: GUI text for pop_spectopo seems to confuse milliseconds/samples

olafdimigen, 2020-06-13:
EEGLAB version 2019_1

For `pop_spectopo`, the GUI asks for` "Epoch time range to analyze **[min_ms max_ms]**` 

This seems to be incorrect - the GUI appears to expect the sample range rather than ms range. The sample range, that is,  `[0  size(EEG.data,2]`, is also entered per default. 

Since some users may intend to only analyze certain parts/minutes of their dataset here, I'd consider this bug somewhat important.

arnodelorme, 2020-06-16:
Seems to be working as expected for me.

arnodelorme, 2020-06-16:
[image]

arnodelorme, 2020-06-16:
And history is

figure; pop_spectopo(EEG, 1, [0 238304.6875], 'EEG' , 'percent', 15, 'freq', [6 10 22], 'freqrange',[2 25],'electrodes','off');

arnodelorme, 2020-06-16:
All in millisecond. I am closing this report, but please reopen it if you judge the issue has not been addressed. Thank you for taking the time.

closed 2020-06-16 by arnodelorme

olafdimigen, 2020-06-16:
You are right - sorry about that one. There's indeed no issue here. (Analyzed a dataset sampled at 1000 Hz, rather than my usual 500 Hz, that where the confusion came from).

arnodelorme, 2020-06-16:
Thank you Olaf