sccn/eeglab #176: Unintialised variable 'opt' in pop_erpimage when calling with no varargin

zenBen, 2020-06-26:
### Description

When calling pop_erpimage with no varargin, function hits line 668 with empty 'options' variable and thus enters block (line 669-680) which expects variable 'opt' to exist, but it doesn't.

<br>
#### Steps to Reproduce

1. Load valid data
2. Call pop_erpimage(EEG, 1, 1)

<br>
#### Expected behavior:
Plot of erpimage for given dataset and channel

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |

arnodelorme, 2020-07-06:
We have made a change to the function #bf65d3cb6. Would you mind to let us know if that works for you?

Arno

arnodelorme, 2020-07-06:
The corrected function is available at 

https://github.com/sccn/eeglab/blob/develop/functions/popfunc/pop_erpimage.m

Please reopen the bug report if this does not fully fix the issue for you.

closed 2020-07-06 by arnodelorme