sccn/eeglab #177: Unrecognizable function or variable 'hdl'

ilaria-b, 2020-06-28:
ERSP analyses:
After precomputing channel measures for ERSPs and ITCs, I opened the plot channel measure GUI, selected my stats, parameters and channel preferences when an error as follows appears:

EEGLAB error in function std_plot() at line 298:

Unrecognized function or variable 'hdl'.

Everything worked the week before, I reran the precomputing channel measures via the GUI, in the study info I specified a separate folder for the output as done in prior occasions. This time error appears. Turned off and on the computer with no improvement.

only difference, there could have been some automatic updates installed when turning on the computer.

| OS version     | [Windows 10 Home (18362.900)] |
| Matlab version | [R2020a] |
| EEGLAB version | [14_1_2b] |

ilaria-b, 2020-06-28:
[image]

arnodelorme, 2020-07-06:
This is referring to a version of EEGLAB (14_1_2b) that is no longer supported. It must be because of plotting options. This line is about color bar. You can safely comment it.

closed 2020-07-06 by arnodelorme