sccn/eeglab #178: Impossible to run a decomposition by ICA using runica

TristanVenot, 2020-07-01:
### Description

I use neuro electrics Enobio files (Easy) that can be imported into a data set. I cannot run a decomposition with normal ICA (runica) however Jader decomposition works. When I have a study I cannot perform the preclustering because it says that the ICA was not run however it says that it is ready to precluster in the starting panel.

https://drive.google.com/drive/folders/1_os1CRNXEQK6eF8PqxqkrlJET40gMmc8?usp=sharing

#### Steps to Reproduce

1. Load the data set
2. tools->decompose by ICA with runica
3. With a study, run PCA -> build Preclustering array with scalp maps

#### Versions

|   |   |
|----| ---- | 
| OS version     | Windows 10|
| Matlab version | 2019a|
| EEGLAB version |v2019.1 |

arnodelorme, 2020-07-07:
Dear Tristan, Your data has huge offset. Remove the offset using menu item "Tools > Remove epoch baseline" then you can run any ICA algorithm (I have just tried). Cheers, Arno

TristanVenot, 2020-07-08:
Hi Arno,

thank you for your answer, it works ! I had another issue, I am trying to perform a statistical analysis and I keep getting this and I do no know from where it comes from. I am sorry I am new to eeglab and I am not quite sure of everything... 
I do the precompute channel measures with ERPS and then when I want to plot it says : 
[image]

Cheers,
Tristan

arnodelorme, 2020-07-08:
Maybe you are trying to plot measure you have not precomputed?

TristanVenot, 2020-07-09:
I precomputed them with ERSP but in the matlab console it says :
.Computing time/frequency decomposition...
Cound not start parallel job
Done.

what does that mean ?

Cheers,
Tristan

arnodelorme, 2020-07-09:
Would you mind to try to download the EEGLAB zip file available on Github?

arnodelorme, 2020-07-10:
Actually, the message is misleading. The ERSP was computed, just not using parallel computation.

closed 2020-07-10 by arnodelorme

TristanVenot, 2020-07-10:
Okay ! Well I tried with the zip file on github bu it presents the same result...