sccn/eeglab #180: std_erspplot bugs/inconsistencies and inflexibility with baseline subtraction

bfbarry, 2020-07-03:
### Description

(This is a combination of issues, followed by a request; but are all under the same theme)

Bugs: 
- When newtimefbaseln.m is called in std_erspplot.m, it handles `trialbase` incorrectly (or at least differently than newtimef.m does).  
- Inefficiency: in newtimeftrialbaseln.m, there are redundant booleans for checking if  `trialbase` is on or off (lines 94, 97).
- Typo: the description for std_erspplot's other optional inputs describes optional inputs for `pop_specparams()` but this should say `pop_erspparams()`

Request: std_erspplot does not offer a chance to change the baseline when calling it.  The setting for `trialbase` is also immutable.  It would be useful to be able to manipulate these parameters for data analysis.

<br>

#### Steps to Reproduce

1. Just investigating code of baseline subtraction within std_erspplot.m upon calling `std_erspplot(STUDY, ALLEEG, 'clusters', clusters, 'plotconditions','together','plotmode')`

<br>

#### Expected behavior:

We expected there to be a way to change the baseline when calling `std_erspplot`.

<br>

#### Actual behavior:
We were not able to change the baseline and found some inconsistencies between functions used by std_erspplot.m.

<br>

#### Versions

|   |   |
|----| ---- | 
| OS version     | X2go |
| Matlab version | 2017a |
| EEGLAB version | 2019.1 |

jiversen, 2020-07-03:
Regarding the first bullet:

`newtimef` & `newtimeftrialbaseln` treat `'trialbase', 'off'` as 'do nothing'.

In contrast, `newtimefbaseln` treats the default (`'trialbase', 'off'`) the same as '`trialbase','full'`, and does a baseline correction, while it does no baseline correction for `'trialbase','on'`. That seems incorrect, or at the least is misusing the meaning of the settings.

Correction: all of the baseline functions should work the same, or some explanation added why not.

Thanks

arnodelorme, 2020-07-06:
These are 2 different functions. One is taking care of the baseline at the trial level (newtimeftrialbaseln) and one is taking care of the baseline after averaging trials (newtimefbaseln) across conditions. 

When we have 'trialbase', 'on', no further baseline (after averaging) is necessary or should be applied. However, when 'trialbase' is not 'on', the baseline depends on whether baseline is NaN or not. I agree that this combination of parameter is not intuitive.

closed 2020-07-06 by arnodelorme

jiversen, 2020-07-10:
Thanks, Arno. Ok, It's a bit confusing, but we'll manage :)

To make sure we understand, can you explain 'trialbase', 'full' in this context?

arnodelorme, 2020-07-10:
Yes,  'trialbase', 'full' will consider the full trial length for baseline at the single trial level. However, after this, you can still apply a pre-stimulus baseline after averaging trials. This is described in 

https://doi.org/10.3389/fpsyg.2011.00236