sccn/eeglab #181: Alois code update

arnodelorme, 2020-07-14:
Dear Arno,

attached are two patches for Eegkab (against commit f2fac387a6d35)

The first was necessary to make eeglab usable on octave 4.4 and 5.2.

The second patch enables the use mexbiosig (mexSLOAD/mexSOPEN) as a
possible alternative to biosig4octmat (sopen/sread/sclose/sload). This
has some improvments in terms of performance (performance, feature
support). Most notable it also improves support of EDF+ format
(multi-channel annotations, sub-second accuracy of timestamps).

Please consider these two patches for inclusion in EEGlab.

Best regards,
  Alois

arnodelorme, 2020-07-14:
[Archive 2.zip](https://github.com/sccn/eeglab/files/4921496/Archive.2.zip)

amisepa, 2021-03-11:
@arnodelorme Was this implemented? Because it may have the solutions to most issues we are having with annotations/events from EDF+ data. I looked at the files and have not idea how to implement it.

amisepa, 2021-03-11:
I have tried to implement eeglab-0002-add-mexbiosig-support.diff by copy-pasting the changes, but it's not working.
[pop_biosig.zip](https://github.com/sccn/eeglab/files/6126255/pop_biosig.zip)

arnodelorme, 2021-03-11:
Thanks for your effort. Let me know when it does work.

arnodelorme, 2021-03-16:
@amisepa this is the modified file

arnodelorme, 2021-03-16:
[pop_biosig.zip](https://github.com/sccn/eeglab/files/6152731/pop_biosig.zip)

arnodelorme, 2021-03-16:
Use MELD or other tools to visualise changes https://www.git-tower.com/blog/diff-tools-windows/

amisepa, 2021-03-17:
Done. I sent a pull request. It is working on my side.

closed 2021-08-30 by arnodelorme