sccn/eeglab #184: Error using chol "Matrix must be square" when std_stat.m with 'fieldtripmethod','montecarlo'

CorentinWicht, 2020-07-24:
### Description
Dear Prof. Delorme,

I am reporting a bug issued by the **std_stat.m** function when trying to run statistics using **'montecarlo' permutations**.

I am using the example provided in the function description (just replaced the [800x12] by random numbers):

**data = { rand(800,10) rand(800,10) rand(800,10);rand(800,10) rand(800,10) rand(800,10)};  % 3 groups, cond 2
[pcond, pgroup, pinter, statscond, statsgroup, statsinter] = ...
 std_stat(data, 'groupstats', 'on','fieldtripmethod','montecarlo','fieldtripmcorrect','max','mode','fieldtrip');**

Here is the error that I get:
_Error using chol
Matrix must be square.

Error in montecarlo (line 13)
R = chol(Sigma);   % Y = R^{-T}X has covariance I

Error in ft_freqstatistics (line 196)
  [stat] = statmethod(cfg, dat, design);

Error in statcondfieldtrip (line 226)
            stat            = ft_freqstatistics(cfg, newdata{:});

Error in std_stat (line 202)
            [F df pval]   = statcondfieldtrip(data(c,:), 'paired', opt.paired{2}, params{:});_

Do you have any idea why I get this error? 
Did you update this function since the EEGLAB v.14.1.2b version?

Best regards,

Corentin

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*Microsoft Windows 10 Enterprise v.10.0.17763*] |
| Matlab version | [*MATLAB R2018b*] |
| EEGLAB version | [EEGLAB v14.1.2b] |

arnodelorme, 2020-08-14:
You should try with EEGLAB 2020.0 (available online). I run your code and it worked as expected.

closed 2020-08-14 by arnodelorme

CorentinWicht, 2020-08-20:
Actually when running the lines of code in EEGLAB v2020.0 the same error is returned:

```
>> data = { rand(800,10) rand(800,10) rand(800,10);rand(800,10) rand(800,10) rand(800,10)}; % 3 groups, cond 2
[pcond, pgroup, pinter, statscond, statsgroup, statsinter] = ...
std_stat(data, 'groupstats', 'on','fieldtripmethod','montecarlo','fieldtripmcorrect','max','mode','fieldtrip');

1 x 3, unpaired data, computing F values
the call to "ft_selectdata" took 0 seconds
using "montecarlo" for the statistical testing

Error using chol
Matrix must be square.

Error in montecarlo (line 13)
R = chol(Sigma);   % Y = R^{-T}X has covariance I

Error in ft_freqstatistics (line 196)
  [stat] = statmethod(cfg, dat, design);

Error in statcondfieldtrip (line 237)
            stat            = ft_freqstatistics(cfg, newdata{:});

Error in std_stat (line 236)
            [F, df, pval]   = statcondfieldtrip(data(c,:), 'paired', opt.paired{2}, params{:});
```

#### Versions
| OS version | [Microsoft Windows 10 Enterprise build 17763]| 
Matlab version | [MATLAB R2019b] | 
EEGLAB version | [EEGLAB v2020.0] |