sccn/eeglab #185: Trouble importing ICA weights using pop_editset function

HelenaSchuch, 2020-07-28:
When importing ICA weights from a text file generated in a previous session of EEGlab using pop_editset it is expected something like:
EEG = pop_editset(EEG, 'run', [ ], 'icachansind', 1:EEG.nbchan, 'icasphere', 'eye(30)', 'icaweights', [filepath, filesep, 'filename.txt']);

But actually the function is generating the output as:
EEG = pop_editset(EEG, 'run', [ ], 'icachansind', 'eye(30)', 'icasphere', [filepath, filesep, 'filename.txt']);

This issue is present both using EEGlab interface and the corresponding command indicated in "ALLCOM".

arnodelorme, 2020-07-28:
Dear Helena, I was not able to reproduce the issue. Created a weight file and imported it. There is no 'icachansind' in the history. If I set 'icachansind' manually, then it is the same in history.

Can you give me more details on how to reproduce the issue?

HelenaSchuch, 2020-07-30:
Dear Arno, thank you for the quick reply. 
I am working from a dataset that is already merged for multiple trial blocks and re-referenced. In the process of merging datasets I did not preserve ica weights from a specific dataset, so 'icachansind' refers to a field in my EEG structure, together with 'icaweights' and 'icasphere'. 
[image]

When I tried to use the function pop_editset to load icaweights to this merged dataset, I received the following warning: "weights matrix cannot be empty if sphere matrix is not, correcting..."
[image]

After that, a 30x30 identity matrix was loaded to 'icachansind' (eye30) - I expected a 1x30 matrix instead, of channel indices from 1 to 30 (which can be obtained from EEG.nbchan). The other ica fields in the EEG struct remained empty. So I guess what is really happening is that the variables are being addressed in the wrong order.
[image]

In conclusion, the identity matrix that should be loaded to 'icasphere' is going to 'icachansind' and the function attempts to load the ica weight matrix to 'icasphere', which leads to the warning because there's no remaining data to be addressed to 'icaweights'.

If you wish to reproduce exactly what I have done, I created a shareable link with one exemplary of my datasets and the corresponding ica weight matrix .txt file
https://drive.google.com/file/d/1Yo2WzjJ7_FYhaF97vQkVVB0xq1IZ-jX_/view?usp=sharing

The commands used were: 
[image]

cll008, 2021-05-27:
Hi @HelenaSchuch, is this still an issue? I can take a look if you share your example files again. I'll close the issue for now but please share and reopen if you have a new link. Thanks.

closed 2021-05-27 by cll008