sccn/eeglab #186: pop_resample did not work due to matrix dimension issue

zhang470, 2020-07-31:
### Description

pop_resample(EEG,250);

Error using  * 
Incorrect dimensions for matrix multiplication. Check that the number of columns in the
first matrix matches the number of rows in the second matrix. To perform elementwise
multiplication, use '.*'.

Error in pop_resample>myresample (line 300)
    b = p * b; % Normalize filter kernel to inserted zeros

Error in pop_resample (line 159)
            tmpres  = [ tmpres; myresample( double( sigtmp(bounds(ind):bounds(ind+1)-1,:)),
            p, q, usesigproc, fc, df ) ];

This problem occurred before and was fixed in EEGLAB Version 14. But the problem showed up again in the EEGLAB v2019.1 

<br>

#### Versions

|   |   |
|----| ---- | 
| OS version     | [*Win10*] |
| Matlab version | [*2020a*] |
| EEGLAB version | [*2019.1*] |

zhang470, 2020-07-31:
Matlab 2016b also had this issue with pop_resample.m.

arnodelorme, 2020-07-31:
Would you mind to share the dataset you are using?

zhang470, 2020-07-31:
I tried my own data recorded with 64 channel ANT system, EEG data recorded with Neuroscan with 67 channels, and the default .set files with 32 channels ('eeglab_data.set' and  'eeglab_data_epochs_ica.set') in the sample_data folder that are included in the EEGLAB package. They all had same problem with pop_resample. I tried different resample rates and it showed the same error message for all the EEG files (raw and epoched) regardless of the EEG system.

zhang470, 2020-07-31:
pop_resample in the earlier EEGLAB version 14 worked fine with all these EEG files in Matlab 2016b and 2020a.

arnodelorme, 2020-07-31:
I have just tried with the sample dataset and I do not have this problem, not has it been reported before, which seems to point toward you own configuration.

Would you mind to type

which resample
which firws

Best wishes,

Arno

zhang470, 2020-07-31:
which firws
C:\toolbox\eeglab\plugins\firfilt2.4\firws.m
>> which resample
C:\Program Files\MATLAB\R2020a\toolbox\signal\signal\resample.m

I tested the EEGLAB development version that I used last year and there was no problem with pop_resample with the same firws.m and resample.m in my setup. I even replaced the pop_resample with the newer on in EEGLAB v2019.1. The new pop_resample.m also worked fine with the 2018 version of EEGLAB. But somehow the same file in EEGLAB v2019.1 does not work.

arnodelorme, 2020-07-31:
pop_resample is identical in the development version and EEGLAB 2019.1 (I have just checked) so it cannot be a problem with this function.
It is probably another function.
P is a scalar (number), there is no reason why the multiplication b * p should fail.

cll008, 2021-05-27:
Closing the issue since it has been inactive, but please feel free to reopen. Thanks.

closed 2021-05-27 by cll008