sccn/eeglab #188: EEG.event.duration wrong in epoched files

alenarto, 2020-08-03:
### Description

In an epoched file, I compared the EEG.epoch(x).eventduration(y) value with EEG.event(EEG.epoch(x).event(y)).duration fields. The value in the EEG.event struct is wrong, the value in the EEG.epoch struct is right. So during epoching, something distorts the EEG.event elements. Other fields are correct, only the duration field is wrong.

<br>
#### Steps to Reproduce

1. Construct an epoched file.
2. Compare duration field in epoch versus event structs. 

<br>
#### Expected behavior:
The same value in EEG.event.duration vs EEG.epoch.eventduration(x).

<br>
#### Actual behavior:
Eg., for event at location 11 (corresponding to latency=0) in first epoch

>> {EEG.event(11).duration EEG.epoch(1).eventduration{11}}

ans =

  1×2 cell array

    [500.5681]    [2.0023e+03]

Note the mismatch. The left hand value is erroneous. The right hand value is correct.

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | n/a on cluster |
| Matlab version | 9.1.0.441655/R2016b |
| EEGLAB version | 14.1.2 |

arnodelorme, 2020-08-03:
Hi Alejandra, Can you provide an example dataset and the command you used?

alenarto, 2020-08-15:
@arnodelorme 

Attached is a sample file and below are commands that show you the error. The epoched file was created using pop_epoch.m. The source continuous file durations agree with durations in the epoch-struct (EEG.epoch) of epoched file but not durations in event-struct (EEG.event) of epoched file. Agatha

###########

    % find onset event for epoch1
    onsets = [EEG.epoch(1).eventlatency{:}];

    %  find location of stimlocked event
    stimloc  = find(onsets == 0);

    % look at event duration for this event saved in the epoch struct
    EEG.epoch(1).eventduration(stimloc)

    ans =

      1×1 cell array

        {[2.0015e+03]} %correctly shows 2 seconds
 
    % now look at the duration for this event, that is
     %saved in the event struct
    EEG.event(EEG.epoch(1).event(stimloc)).duration

    ans =

      500.3721 %incorrectly shows 500 ms

FILE: [sample7epochs1channel.zip](https://github.com/sccn/eeglab/files/5079508/sample7epochs1channel.zip)

alenarto, 2020-08-15:
Sorry @arnodelorme accidentally clicked close rather than commend! See notes above.

cll008, 2021-05-27:
Hi Agatha! Could you also share the continuous set file? Looks like somehow event duration has been quartered. I wonder if this complication comes from resampling or some srate issues.

alenarto, 2021-06-03:
We do downsample so that’s possible. Remind which file did I send you? I’ve had a computer crash and had to get a new computer so I may have to recreate some of this :-/

cll008, 2021-06-09:
Hm, it's named sample7epochs1channel.set, but it looks like it was named sub-1070302_ses-01_task-eoec_eeg

alenarto, 2021-06-15:
Sorry for the delay - let me pull that original file for you.

alenarto, 2021-06-22:
Here you go - [full file](https://www.dropbox.com/s/t692udikqydgzcj/sub1070302_full.zip?dl=0).

cll008, 2021-06-25:
Thanks Agatha... Looks like the issue really is in the downsampling. EEG.event.duration is reduced whereas EEG.event.epochduration is not.  The problem is in pop _resample https://github.com/sccn/eeglab/blob/develop/functions/popfunc/pop_resample.m#L200

@arnodelorme Event `duration` is expressed in ms (not data points) right? Same situation for `latency` right? Looks like those lines should be removed?

cll008, 2021-07-01:
Hi Agatha, so EEG.event.duration is apparently stored as data samples but displayed in seconds for continuous data, and in milliseconds for epoched data. 

I'll double check EEG.epoch.eventduration. Anyways I think the definition should be consistent or the naming convention should be different.

cll008, 2021-07-03:
https://github.com/sccn/eeglab/blob/0ae1baf0e92fb18ab1af2de6dddfe93574d480fd/functions/adminfunc/eeg_checkset.m#L624

`EEG.epoch.eventduration` is indeed expressed in ms instead of frames. @arnodelorme have a few options here

1. Leave things as they are but update EEG.epoch documentation
2. Change to frames (and include `EEG.epoch.eventduration` in pop_resample). 
3. Change to seconds (instead of ms) to match BIDS spec event.duration
2 and 3 are probably complicated depending on what is downstream of EEG.epoch.eventduration. Not sure how this field is usually used.

arnodelorme, 2021-07-06:
Update documentation because eventduration is consistent with eventlatency in ms as well.

cll008, 2021-07-07:
Done https://github.com/sccn/sccn.github.io/commit/d52d258ed7824928996e432167477edfe4aa863e

closed 2021-07-07 by cll008