sccn/eeglab #190: pop_comperp.m missing from GUI

schalles, 2020-08-05:
### Description

Sum/Compare ERPs used to be a GUI menu item in v 14, but is missing in most recent version - it is still present for comparing component space ERPs, but not for sensor space.

| OS version     | Win10/xUbuntu 20.04|
| Matlab version |R2015|
| EEGLAB version | v2019.1 |

arnodelorme, 2020-08-14:
Yes, it is still here. You need to go in the preferences and activate the old menus. This menu is obsolete because everything can be done better in studies.

closed 2020-08-14 by arnodelorme

schalles, 2020-08-14:
Ah, thanks Arno.

I stopped using the study structure when I transitioned from human testing
in groups and now do almost exclusively single subject experiments with
dolphins.

Matt Schalles, Ph.D.
Research Associate
Auditory Neuroscience Lab, BU

arnodelorme, 2020-08-14:
Sounds like fun :-)