sccn/eeglab #192: problem with importing .eeg format

amirmoeini1993, 2020-08-08:
### Description

https://www.dropbox.com/s/fdsxxpbuct3l42q/sub-S01.rar?dl=0

<br>
#### Steps to Reproduce

I try to load .eeg format with MatLab 
loadeeg('')
and with eeglab 
file>>>import data>>>use EEGLab functions and plugins>>>from neuroscan .EEG

<br>
#### Expected behavior:
loading data

<br>
#### Actual behavior:
I have a problem with importing .eeg format in EEG Lab although my data size is not so huge, it gives me an error at line 177 out of memory. I try with two different versions so I  want to make sure that data has no problem.

<br>
#### Versions
Windows10
matlab 2013a  matlab 2018b
EEGLAB 2013a EEGLAB 2019.0

arnodelorme, 2020-08-14:
This is not a neuroscan file. You need to use the File-IO plugin to import this file (I did it successfully).

closed 2020-08-14 by arnodelorme

arnodelorme, 2020-08-14:
@widmann I fixed an issue associated with reading that data (that reads OK under File-IO but made BVA-IO crash). We should issue a new release of BVA-IO (but see also my other comment).

widmann, 2020-08-17:
Arno, sorry, I disagree with this change. The provided file is not standard conforming. The problematic line (9; "Data orientation: MULTIPLEXED=ch1,pt1, ch2,pt1 ...") should actually be commented/is a comment. If you prefix this line with a semicolon the file is loaded flawlessly.

Please revert the change. Thanks!

The bug presumably is in the MNE Analyzer export.

arnodelorme, 2020-08-17:
OK, I agree with you. 

Could we though simply skip the line and show a message in red that the file is badly formed and might be corrupted and still import it. This is going to be a blocker for a lot of people. These files are BIDS files that are publicly released. And in BIDS datasets, they have to change not one but dozens of files manually.

Arno

widmann, 2020-08-19:
Implemented as suggested.

Best,
Andreas

arnodelorme, 2020-08-19:
Hi Andreas,

Are we ready to release a new revision of the new plugin?

Arno

widmann, 2020-08-19:
No, I would like to wait for the revision of the ahdr import. See other msg.
Andreas

arnodelorme, 2020-08-19:
OK. Once it is done, this is the link to upload a plugin revision to the plugin manager

http://sccn.ucsd.edu/eeglab/plugin_uploader/version_update.php

Or I can do it if you prefer.
Best wishes,

Arno