sccn/eeglab #193: BCI2000 plugin: Error to  edit channel location file.

Candidasfb, 2020-08-10:
### Description

I am using the BCI2000 plugin to open a .DAT eeg file. When I try to upload the channel location file using the Edit->channel location, I got the following error message 

_Reference to non-existent field 'X'.

Error in eeg_checkchanlocs (line 183)
if (any( cellfun('isempty',{ chanedit.X })) || any( cellfun('isempty', { chanedit.theta})) || any( cellfun('isempty', { chanedit.sph_theta}))) &&...

Error in pop_chanedit (line 217)
    [tmp chaninfo chans] = eeg_checkchanlocs(chans, chaninfo);
 
Error while evaluating Menu Callback.**__

Any thoughts about what is going on?

Thank you in advance.

<br>
#### Steps to Reproduce

1.File -> Import data -> Using EEGLAB functions and plugins -> from BCI2000.DAT file
2. Edit ->channel location

<br>
#### Expected behavior:
pop up a window to upload the channel location file.

<br>
#### Actual behavior:
The following error 

Reference to non-existent field 'X'.

Error in eeg_checkchanlocs (line 183)
if (any( cellfun('isempty',{ chanedit.X })) || any( cellfun('isempty', { chanedit.theta})) || any(
cellfun('isempty', { chanedit.sph_theta}))) &&...

Error in pop_chanedit (line 217)
    [tmp chaninfo chans] = eeg_checkchanlocs(chans, chaninfo);
 
Error while evaluating Menu Callback.

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*windows 10 home*] |
| Matlab version | [*R2019b*] |
| EEGLAB version | [*EEGLAv2019.1*] |

arnodelorme, 2020-08-14:
Would you mind to share a datafile? 

@cboulay you wrote that plugin. What do you think?

Candidasfb, 2020-08-15:
Thank you for your reply. 

Please, find the file attached. 

[16-chan_created.zip](https://github.com/sccn/eeglab/files/5078691/16-chan_created.zip)

Daisy-Ding1124, 2020-12-07:
I'm facing the same problem. [OS: macOS Big Sur 11.0.1; Matlab version: R2020a; EEGLAB version: v2019.1] Has you found the reason now? I'd appreciate it a lot if you can share your solution.

Candidasfb, 2020-12-07:
Unfortunately, I did not have any solution. Please, let me know if you get it!

Thanks

cll008, 2021-05-27:
Looks like when the BCI2000 import plugin is used `EEG.chanlocs` is 0x1 struct with a `labels` field and EEGLAB gets confused. Simply resetting the strucutre in the command line by entering `EEG.chanlocs = []` before clicking Edit -> Channel Location works.

closed 2021-05-27 by cll008