sccn/eeglab #196: Silent error handling parameters in imagesclogy.m

zenBen, 2020-08-14:
### Description

Function ignores color space limits if the number of parameters exceeds 4, i.e. if called with `xticks` or `yticks` parameters.
See lines 77 - 81
```
  if nargin == 4 && ~isempty(clim)
      imagesc(times,newfreqs,data,clim);
  else 
      imagesc(times,newfreqs,data);
 end
```
inputParser should be preferred!

**Additionally**, if user wants to input varargin, they will bypass built-in code to discover appropriate spacing for yticks, see lines 87 - 93:
 ```
 if nargin >= 6
      divs = yticks;
  else 
      divs = linspace(log(freqs(1)), log(freqs(end)), 10);
      divs = ceil(exp(divs)); divs = unique_bc(divs); % ceil is critical here, round might misalign
                                               % out-of border label with within border ticks
  end
```
This will throw following error if user does not define their own yticks (expecting it to be found by the function as normal).
```
Error using matlab.graphics.axis.Axes/set
Value must be a vector of type single or double whose values increase

Error in imagesclogy (line 95)
  set(gca, 'ytick', divs);
```

<br>
#### Steps to Reproduce

1. call imagesclogy.m with 4 parameters
2. call imagesclogy.m with 5+ parameters
3. call imagesclogy.m with 5 parameters plus any varargin name, value pairs

<br>
#### Expected behavior:
Get an image

<br>
#### Actual behavior:
Get color-inconsistent images, or crashing

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | Ubuntu 18.x |
| Matlab version | R2019b |
| EEGLAB version | 2019 (from repo) |

arnodelorme, 2020-08-14:
Hi Ben,

Would you mind to send an updated version of this function that can handle these changes?
Best wishes,

Arno

closed 2020-08-14 by arnodelorme

zenBen, 2020-08-17:
Benjamin Cowley PhD
Docent of Cognitive Science
Associate Professor at Faculty of Education, University of Helsinki, Finland.

zenBen, 2020-08-17:
Updated to return the image handle

Benjamin Cowley PhD
Docent of Cognitive Science
Associate Professor at Faculty of Education, University of Helsinki, Finland.
________________________________
From: Cowley, Ben <ben.cowley@helsinki.fi>
Sent: 17 August 2020 12:41
To: sccn/eeglab <reply@reply.github.com>
Subject: Re: [sccn/eeglab] Silent error handling parameters in imagesclogy.m (#196)

Benjamin Cowley PhD
Docent of Cognitive Science
Associate Professor at Faculty of Education, University of Helsinki, Finland.