sccn/eeglab #198: std_readdata error

CorentinWicht, 2020-08-20:
### Description

Dear EEGLAB,

I am writting here to report a bug related to the std_readdata function (also in the GUI).

<br>
#### Steps to Reproduce

1. [*First Step*]
Load the [example study](https://www.dropbox.com/sh/tlkq03le7qp71oo/AAAA5oZ5q_VBX8a9K5sZRZXoa?dl=0)

2. [*Second Step*]
STUDY --> Precompute channel measures --> select only Power Spectrum with 'specmode', 'psd', 'logtrials', 'off' and untick the spherical interpolation (i.e. no influence on the error)

[image]

3.  [*Third Step*]
STUDY --> Plot channel measures --> Select all channels to plot and ask for spectra and...

[image]

**Hypotheses**
I suspect the error is coming from the fact that my resting-state datasets only have 1 trial/epoch while this seems to be problematic when computing 'psd'.
Indeed, the error disappears when using 'fft' instead but another one comes up:

[image]

**Code to reproduce the error**
```
[STUDY ALLEEG] = pop_loadstudy('filename', 'Study_example.study', 'filepath', 'E:\Dropbox\DOC3\MATLABTests\EXAMPLE');

[STUDY ALLEEG] = std_precomp(STUDY, ALLEEG,'channels','spec','on','recompute','on',...
    'specparams',{'specmode','psd','logtrials','off'});

[STUDY,SpectData,SpectFreqs] =  std_readdata (STUDY, ALLEEG,'channels',...
    {ALLEEG(1).chanlocs.labels},'datatype','spec');

This code was nicely working with EEGLAB 14.1.2b (though I used std_readspec instead of std_readdata) but now crashes with EEGLAB2020.
```

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | Windows 10 Enterprise build 17763 |
| Matlab version | Matlab R2019b |
| EEGLAB version | EEGLAB v2020.0 |

CorentinWicht, 2020-08-20:
Actually when using the code provided abobve and changing 'specmode', 'psd' to 'specmode','fft',

the following error is returned by std_readdata:

```
Unable to perform assignment because the size of the left side is 48-by-833 and the size of the right side is 48-by-833-by-1-by-64.

Error in std_readdata>reorganizedata (line 335)
                    case 3, datavals{iItem}(:,:,count:count+numItems-1) = dataTmp{iCase}{iItem};

Error in std_readdata (line 308)
datavals = reorganizedata(dataTmp, dim);
```

Hence, the problem is not solved by chaging the specmode...

arnodelorme, 2020-08-20:
First error I was not able to reproduce. I followed the instructions. 

[image]

arnodelorme, 2020-08-20:
Hi Corentin, I have fixed the PSD option as well on Github. It is returning virtually identical results compared to FFT (which makes sense) - right is PSD and left is FFT. We will need to figure out why the scale is offset though. I will close this report and submit a new one.

[image]

closed 2020-08-20 by arnodelorme

Naba1210, 2022-10-25:
Description
Dear EEGLAB,

I am writting here to report a bug related to the std_stdcond function (also in the GUI).

Study location: https://drive.google.com/drive/u/2/folders/0ALSJMw_UHlf9Uk9PVA
#### Steps to Reproduce
[First Step]
STUDY-->Run ICA-->Head Model and fitting-->Component Dipole autofit-->DIPFIT<15%

[Second Step]
STUDY --> Precompute component measures --> 
[image]

[Third Step]
STUDY -->Pre clustering
[image]

Hypotheses
I suspect that the data structure has similar events but there are some files which are showing uevents "[]"
The original files are all having same data structure, but turns out after precomputing this was the only difference i could find
[image]

Code to reproduce the error

val =
    'STUDY = [];
     [STUDY ALLEEG] = std_checkset(STUDY, ALLEEG);
     [STUDY ALLEEG] = std_editset( STUDY, ALLEEG, 'name','Objective 1 ERSP','commands',{{'index' 1 'subject' 'S1'} {'index' 2 'subject' 'S2'} {'index' 3 'subject' 'S3'} {'index' 4 'subject' 'S4'} {'index' 5 'subject' 'S5'} {'index' 6 'subject' 'S6'} {'index' 7 'subject' 'S7'} {'index' 8 'subject' 'S8'} {'index' 9 'subject' 'S9'} {'index' 10 'subject' 'S10'} {'index' 1 'condition' 'B1'} {'index' 2 'condition' 'B1'} {'index' 3 'condition' 'B1'} {'index' 4 'condition' 'B1'} {'index' 5 'condition' 'B1'} {'index' 6 'condition' 'B1'} {'index' 7 'condition' 'B1'} {'index' 8 'condition' 'B1'} {'index' 9 'condition' 'B1'} {'index' 10 'condition' 'B1'} {'index' 1 'group' 'a'} {'index' 2 'group' 'a'} {'index' 3 'group' 'a'} {'index' 4 'group' 'a'} {'index' 5 'group' 'a'} {'index' 6 'group' 'a'} {'index' 7 'group' 'a'} {'index' 8 'group' 'a'} {'index' 10 'group' 'a'} {'index' 9 'group' 'a'} {'index' 11 'subject' 'S11'} {'index' 12 'subject' 'S12'} {'index' 13 'subject' 'S13'} {'index' 14 'subject' 'S14'} {'index' 15 'subject' 'S15'} {'index' 16 'subject' 'S16'} {'index' 17 'subject' 'S17'} {'index' 18 'subject' 'S18'} {'index' 19 'subject' 'S19'} {'index' 20 'subject' 'S20'} {'index' 11 'condition' 'B1'} {'index' 12 'condition' 'B1'} {'index' 13 'condition' 'B1'} {'index' 14 'condition' 'B1'} {'index' 15 'condition' 'B1'} {'index' 16 'condition' 'B1'} {'index' 17 'condition' 'B1'} {'index' 18 'condition' 'B1'} {'index' 19 'condition' 'B1'} {'index' 20 'condition' 'B1'} {'index' 11 'group' 'a'} {'index' 12 'group' 'a'} {'index' 13 'group' 'a'} {'index' 14 'group' 'a'} {'index' 15 'group' 'a'} {'index' 16 'group' 'a'} {'index' 17 'group' 'a'} {'index' 18 'group' 'a'} {'index' 19 'group' 'a'} {'index' 20 'group' 'a'} {'index' 21 'subject' 'S21'} {'index' 22 'subject' 'S22'} {'index' 23 'subject' 'S23'} {'index' 24 'subject' 'S24'} {'index' 25 'subject' 'S25'} {'index' 26 'subject' 'S26'} {'index' 27 'subject' 'S27'} {'index' 28 'subject' 'S28'} {'index' 29 'subject' 'S29'} {'index' 30 'subject' 'S30'} {'index' 21 'condition' 'B1'} {'index' 22 'condition' 'B1'} {'index' 23 'condition' 'B1'} {'index' 24 'condition' 'B1'} {'index' 25 'condition' 'B1'} {'index' 26 'condition' 'B1'} {'index' 27 'condition' 'B1'} {'index' 28 'condition' 'B1'} {'index' 29 'condition' 'B1'} {'index' 30 'condition' 'B1'} {'index' 21 'group' 'a'} {'index' 22 'group' 'a'} {'index' 23 'group' 'a'} {'index' 24 'group' 'a'} {'index' 25 'group' 'a'} {'index' 26 'group' 'a'} {'index' 27 'group' 'a'} {'index' 28 'group' 'a'} {'index' 29 'group' 'a'} {'index' 30 'group' 'a'} {'index' 31 'subject' 'S31'} {'index' 32 'subject' 'S32'} {'index' 33 'subject' 'S33'} {'index' 34 'subject' 'S34'} {'index' 35 'subject' 'S35'} {'index' 36 'subject' 'S36'} {'index' 37 'subject' 'S37'} {'index' 38 'subject' 'S38'} {'index' 39 'subject' 'S39'} {'index' 40 'subject' 'S40'} {'index' 31 'condition' 'B1'} {'index' 32 'condition' 'B1'} {'index' 33 'condition' 'B1'} {'index' 34 'condition' 'B1'} {'index' 35 'condition' 'B1'} {'index' 36 'condition' 'B1'} {'index' 37 'condition' 'B1'} {'index' 38 'condition' 'B1'} {'index' 39 'condition' 'B1'} {'index' 40 'condition' 'B1'} {'index' 31 'group' 'a'} {'index' 32 'group' 'a'} {'index' 33 'group' 'a'} {'index' 34 'group' 'a'} {'index' 35 'group' 'a'} {'index' 36 'group' 'a'} {'index' 37 'group' 'a'} {'index' 38 'group' 'a'} {'index' 39 'group' 'a'} {'index' 40 'group' 'a'} {'index' 41 'subject' 'S1'} {'index' 42 'subject' 'S2'} {'index' 43 'subject' 'S3'} {'index' 44 'subject' 'S4'} {'index' 45 'subject' 'S5'} {'index' 46 'subject' 'S6'} {'index' 47 'subject' 'S7'} {'index' 48 'subject' 'S8'} {'index' 49 'subject' 'S9'} {'index' 50 'subject' 'S10'} {'index' 41 'condition' 'B2'} {'index' 42 'condition' 'B2'} {'index' 43 'condition' 'B2'} {'index' 44 'condition' 'B2'} {'index' 45 'condition' 'B2'} {'index' 46 'condition' 'B2'} {'index' 47 'condition' 'B2'} {'index' 48 'condition' 'B2'} {'index' 49 'condition' 'B2'} {'index' 50 'condition' 'B2'} {'index' 41 'group' 'a'} {'index' 42 'group' 'a'} {'index' 43 'group' 'a'} {'index' 44 'group' 'a'} {'index' 45 'group' 'a'} {'index' 46 'group' 'a'} {'index' 47 'group' 'a'} {'index' 48 'group' 'a'} {'index' 49 'group' 'a'} {'index' 50 'group' 'a'} {'index' 51 'condition' 'B2'} {'index' 52 'condition' 'B2'} {'index' 53 'condition' 'B2'} {'index' 54 'condition' 'B2'} {'index' 55 'condition' 'B2'} {'index' 56 'condition' 'B2'} {'index' 57 'condition' 'B2'} {'index' 58 'condition' 'B2'} {'index' 59 'condition' 'B2'} {'index' 60 'condition' 'B2'} {'index' 51 'subject' 'S11'} {'index' 52 'subject' 'S12'} {'index' 53 'subject' 'S13'} {'index' 54 'subject' 'S14'} {'index' 55 'subject' 'S15'} {'index' 56 'subject' [... cut ...]
     STUDY = pop_erspparams(STUDY, 'topotime',NaN,'topofreq',NaN,'timerange',[0 2000] ,'freqrange',[1 25] );
     STUDY = pop_erspparams(STUDY, 'topotime',NaN,'topofreq',NaN,'freqrange',[]);
     [STUDY EEG] = pop_savestudy( STUDY, EEG, 'savemode','resave');
     [STUDY ALLEEG] = std_checkset(STUDY, ALLEEG);
     [STUDY ALLEEG] = pop_loadstudy('filename', 'Objective 1.study', 'filepath', 'D:\');'

ERROR: Number of elements must not change. Use [] as one of the size inputs to automatically calculate the appropriate size for that dimension.