sccn/eeglab #204: std_readdata error: "Cannot handle conditions with different number of components for a given subject"

CorentinWicht, 2020-09-01:
### Description

Dear Prof Delorme,

I am reporting here another bug I recently uncovered in the std_readdata function, this time related to ```ICA components```.
The error is the following:

```
Error using std_readdata (line 290)
Cannot handle conditions with different number of components for a given subject
```

<br>
#### Steps to Reproduce

1. [**First Step**]
Load the [example study](https://www.dropbox.com/sh/tlkq03le7qp71oo/AAAA5oZ5q_VBX8a9K5sZRZXoa?dl=0)

2. [**Second Step: Precompute Components Measure**]

[image]

3. [**Third Step: Build Preclustering Array**]

:
[image]

4. [**Forth Step: Edit/Plot Components Measures**]

Error when trying to ```Plot Spectra```
[image]

** Code to reproduce the error**
```
% Precompute Power Spectra
[STUDY, ALLEEG] = std_precomp(STUDY, ALLEEG, 'components','savetrials','on','allcomps','on',...
    'recompute','on','scalp','on','spec','on','specparams',{'specmode','psd','logtrials','off'});

% Create preclustering array
[STUDY ALLEEG] = std_preclust(STUDY, ALLEEG, 1,{'spec','npca',10,'weight',1,'freqrange',[1 48] },...
    {'scalp','npca',10,'weight',1,'abso',1},{'dipoles','weight',1},{'moments','weight',1});

% Reading data
[STUDY,~,SpectFreqs] =  std_readdata(STUDY, ALLEEG,'component',[],'clusters',1,'datatype','spec');
```

<br>
#### Actual behavior: 

The error seems to be related to lines 286-287 in std_readdata:`

```
compNumbers = cellfun(@(x)size(x, realDim), dataTmp{iDat1});
uniqComps = unique(compNumbers);
```

I am actually unsure why you are doing this? Knowing that dataTmp{iDat1} is (i.e. 1st file in my dataset):

[image]

It doesn't make a lot of sense to me, since the number of components here is actually 25 (i.e. ```size(dataTmp{iDat1}{2},2)```) and I **only have 1 condition for each subject** (i.e. 2 groups between-subject design). 

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | Windows 10 Enterprise build 17763 |
| Matlab version | Matlab R2019b |
| EEGLAB version | EEGLAB v2020.0 |

arnodelorme, 2020-09-02:
I am looking into the problem.

arnodelorme, 2020-09-02:
Hi Corentin, I have implemented fix ee458537e . Let me know if it works for you.

closed 2020-09-02 by arnodelorme