sccn/eeglab #205: Problem with eeglab 2020 and signal processing toolbox

FelixYan101, 2020-09-10:
### Description

I was tring to install matlab and eeglab to a new laptop.
After everything was installed(such as signal processing toolbox and else), when I tried to use eeglab,
" eeglab
Not using signal processing toolbox, if you experience problem, reset your Matlab path to default
Some menus items hidden. Use Preference menu to show them all.
Not using signal processing toolbox, if you experience problem, reset your Matlab path to default"
pops out.

After this If I tried to use such as "cleanline" extension, error message "undefined 'double'........'dpss' " shows up.

<br>
#### Steps to Reproduce

1. [*First Step*] re-install matlab and eeglab to both 2020
2. [*Second Step*] enter eeglab 
3. [*and so on...*] use extension such as cleanline (which needs signal processing toolbox 

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*win10*] |
| Matlab version | [*2020*] |
| EEGLAB version | [*2020a*] |

arnodelorme, 2020-09-11:
Yes cleanline uses the signal processing toolbox. You need to purchase it or use Octave. 

https://sccn.ucsd.edu/wiki/Running_EEGLAB_on_Octave

closed 2020-09-11 by arnodelorme

FelixYan101, 2020-09-11:
Sorry, you might not understand what happened cause I forgot to mention. 
I have matlab and it works normally when I used mac and on my old laptop. (both matlab2019a eeglab2019.0)
It works normally on other api which need signal processing toolbox.
Only on eeglab it shows
"Not using signal processing toolbox, if you experience problem, reset your Matlab path to default"
and that error message.

FelixYan101, 2020-09-11:
The signal processing toolbox was installed and was able to use it when I used other matlab API such as recor.

arnodelorme, 2020-09-11:
I think you must have overloaded the "double" function which convert numbers to double precision. Type "which double" on the matlab command line.

FelixYan101, 2020-09-11:
Here's what I got
>> which double
built-in (E:\Program Files\MATLAB\R2020a\toolbox\matlab\datatypes\double)

By the way, the error message was in Japanese so I simplely translated it 
MATLAB:UndefinedFunction
The function'dpss' corresponding to input parameters of type'double' is not defined.

dpsschk:28
rmlinesmovingwinc:94
cleanline:349
pop_cleanline:50
vykdymas:1113
pushbutton1_Callback:694
gui_mainfcn:95
pop_nuoseklus_apdorojimas:81
@(hObject,eventdata)pop_nuoseklus_apdorojimas('pushbutton1_Callback',hObject,eventdata,guidata(hObject))

arnodelorme, 2020-09-11:
This person had the same problem #153
Can you type "which dpss"

FelixYan101, 2020-09-11:
So I typed which dpss,at first it shows
">> which dpss
E:\Program Files\MATLAB\R2020a\toolbox\signal\signal\dpss.m"

after that, when I opened eeglab, same message
 ">> eeglab
Not using signal processing toolbox, if you experience problem, reset your Matlab path to default
Some menus items hidden. Use Preference menu to show them all."
was given and same error message was given after I tried cleanline.
"Undefined funtion 'dpss' for input arguments of type 'double'."

So I typed which dpss again 
it shows couldn't find dpss.

By the way,the signal processing toolbox was installed and I tried several times to uninstall it and installed again.
Another matlab api I  used"RECOR" also require signal processing toolbox  and it works (different function needed)

arnodelorme, 2020-09-11:
You might have set the option in the Preference not to use the Signal Proc Toolbox. 
I would just delete and reinstall EEGLAB.

Arno