sccn/eeglab #208: Nicolet .e file format read error

janbrogger, 2020-09-17:
### Description
Some Nicolet .e  files fail reading because of a file format problem. This is already fixed in fieldtrip, but the EEGLAB version of fieldtrip is old. 

Issue was posted on fieldtrip https://github.com/fieldtrip/fieldtrip/issues/1548 but then I found out I had already fixed it.

<br>
#### Steps to Reproduce

1.  `pop_fileio('a_failing_file.e')`
```
Reading data ...
Unable to perform assignment because the size of the left side is 1854834-by-1 and the size of the right side is
983352-by-1.
```

Error in read_nervus_data (line 148)
        out(curIdx : end,i) = fread(h, length(out)-curIdx + 1, 'int16') * mult;[*First Step*]

<br>
#### Expected behavior:
No error. File is just read.

<br>
#### Actual behavior:
See above

<br>
#### Versions
|   |   |
|----| ---- | 
| OS version     | Windows 10. |
| Matlab version | 9.5.0.1067069 (R2018b) Update 4 |
| EEGLAB version | EEGLAB is from up-to-date develop branch at https://github.com/sccn/eeglab/tree/73165f45dadb424b8906f99e53774caf8e56d965
 |

janbrogger, 2020-09-17:
Updated fileio to 20200914 which fixed it.

closed 2020-09-17 by janbrogger

arnodelorme, 2020-09-17:
Thanks Jan.