sccn/eeglab #211: Importing channel locations from ANT neuro

lucameroll, 2020-10-14:
Hello everyone!
I am acquiring eeg data with the ANT neuro's 64-channel system and I want to use eeglab to analyse the data. I am using Lab Streaming Layer (LSL) as software for data acquisition, which saves the file in .xdf format. One problem that I am facing is that the electrodes are labelled as numbers, from 1 to 65 (because I have an EOG channel), and I would like them to be labelled following the 10-20 international system (so Fz, Cz, etc), this way the "source" from which the signal comes from will be much more clear. I am not interested in a source estimation, so I don't need anatomical precision, but I want to get the PSD under some electrodes to evaluate cognitive workload.
I have found the label field in the structure dataset, however, I don't know how to import those labels from the xdf dataset to the EEG data structure created in eeglab (because this is what you're supposed to do right?), to then read them with the eeglab GUI. 
Unfortunately on other websites I could not find any helpful solution. Can somebody help me out, please?
Thank you in advance!
Luca

arnodelorme, 2020-10-15:
Hi Luca,

If you know the correspondance
1 -> Cz
2 -> P3
...

Then simply change the labels in the channel editor after importing the file (menu item Edit > Channel locations). When you are done, select the button "Look up locs." to get the template localization for these electrodes. Save the dataset as you can reuse the location for other subjects (menu item Edit > Dataset info, then select channel location from other dataset).

closed 2020-10-15 by arnodelorme

yst19941129, 2022-06-30:
> 

Did you solve this problem please?