sccn/eeglab #212: Bug for synchronizing EEG data and ET data,

luofeifan0311, 2020-10-16:
### Description

When I use EYE-EEG toolbox to synchronize EEG data and ET data, It will report "Sample time in the eye tracking data is not monotonically increasing.Check for backward jumps in the timestamp colum of your raw data and report this to your ET manufacturer"

This is ET file and EEG file
<br>
#### Steps to Reproduce

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |

behinger, 2020-12-01:
I think you should report this here: https://github.com/olafdimigen/eye-eeg

Also your report is missing lots of information right now (no attached files for instance). I would also recommend to provide the code you use to run it.

closed 2021-05-28 by cll008

olafdimigen, 2024-08-31:
This is an old issue from 2020, but maybe someone else stumbles upon it:

This issue arises when the sample time stamps in the eye-tracker (the system time in milliseconds or microseconds assigned to each gaze sample) make a backwards jump in time in the eye-tracking raw data, like this:

231001,  gaze x,  gaze y, etc
231002,  gaze x,  gaze y, etc
231003,  gaze x,  gaze y, etc
23988,  gaze x,  gaze y, etc  # backwards jump!
23989,  gaze x,  gaze y, etc
23909,  gaze x,  gaze y, etc

This happened, for example, with some buggy old Sensomotor Instruments (SMI) eye-trackers. Because we cannot know how much time has really passed during this backward jump, EYE-EEG cannot synchronize the time series with the EEG. With newer firmware version of SMI software, this issue did not occur anymore.