sccn/eeglab #215: Error in Topoplot Function

Prithila05, 2020-10-30:
### Description

I tried to get the topoplots for DEAP dataset for frequency range 0 to 256 at 1Hz interval. I get the error below, however, a smaller frequency range works just fine.

[image]

Can anyone please help me with this?

<br>

|   |   |
|----| ---- | 
| OS version     | [*windows 10*] |
| Matlab version | [*R2019b*] |
| EEGLAB version | [*v2020.0*] |

arnodelorme, 2020-10-30:
This looks like you have not imported channel locations.

Prithila05, 2020-10-30:
Hi,

Thank you for replying. For channel locations I followed the steps:
1. edit
2. channel locations
3. Use MNI coordinates for the BEM Dipfit model

[image]

I get these results for a smaller frequency interval.

Thank you

closed 2020-12-02 by arnodelorme