sccn/eeglab #216: Save Topoplots as image files

Prithila05, 2020-10-30:
Hi,

I wanted to save individual topoplot as image files something like this:

[image]

I want to save several topoplot images for a wide frequency range. Can you please guide me on how to do that? 

Thank you!!

|   |   |
|----| ---- | 
| OS version     | [*windows 10*] |
| Matlab version | [*R2019b*] |
| EEGLAB version | [*v2020.0*] |

arnodelorme, 2020-10-30:
This is a duplicate report of #215

closed 2020-10-30 by arnodelorme

Prithila05, 2020-10-31:
Sorry These issues are not same

arnodelorme, 2020-10-31:
You should direct your question to the eeglablist. This web site is for submission of bugs. Cheers.