sccn/eeglab #217: Issue running AMICA on EEGLAB (runamica15)

AndersonAAU, 2020-11-03:
Hi,
I’m not being able to run the AMICA code using the EEGLAB GUI. I reach the pop_out window for the parameters and when I accept the parameters I get the error:

=======================

Found datfile
Error using fread
Invalid file identifier. Use fopen to generate a valid file identifier.

Error in runamica15 (line 743)
        dattmp = fread(fid,[chans inf],'float');

Error in pop_runamica (line 239)
    [W,S,mods] = runamica15(datfile,arglist{:});

Error while evaluating Menu Callback

=======================

The dataset I am using in the GUI is also saved in the PC disk (as a .fdt/.set file), but it seems the code has trouble loading it… So I need some help to solve this in the most appropriated way.
I’m using Matlab 2019b, and EEGLAB v2019.1 with the latest version of the AMICA plugin.

Thanks in advance for the help.
Best regards,
Anderson

behinger, 2020-12-01:
Hi!
Thanks for reporting. Can you share the dataset / a small version of it? Else it is hard to test.

AndersonAAU, 2020-12-01:
Hi,
Is it OK If I share the EEG workspace that got the error?
I think I can also share the EEGLAB and AMICA folders, but I need to know if you actually need it.
Best regards,
Anderson

behinger, 2020-12-01:
Can you first share what is in "EEG.datfile" and can you confirm that file exists at that location?

arnodelorme, 2020-12-02:
Copying @japalmer29 . This is a bug for Amica rather than EEGLAB so would be better suited for Amica Gihub issue

AndersonAAU, 2020-12-03:
Hi,
I think I solved the problem… my EEGLAB folder was located in my OneDrive directory and this was somehow interfering with the code.
I can re-create the issue if you think this is relevant. Otherwise I can also remove/close the issue in GitHub
Best regards,
Anderson

arnodelorme, 2020-12-03:
Good to know. Thanks for letting us know

closed 2020-12-03 by arnodelorme