sccn/eeglab #222: Starting eeglab produces a warning when checking updates

miamaria27, 2020-11-21:
### Description

When I start eeglab with the command "eeglab" I get the warning:
"Warning: Out of range or non-integer values truncated during conversion to character.
(Type "warning off MATLAB:nonIntegerTruncatedInConversionToChar" to suppress this warning.)
 
> In eeglab at 1154 
Could not check for the latest EEGLAB version (internet may be disconnected).  "

Note that my internet is connected. Since in the line 1154 of eeglab it says "eeglab_update;" there is probably a problem when trying to find out the newest eeglab version. However, when typing the command eeglab_update, it only says ">> eeglab_update;
Could not check for the latest EEGLAB version (internet may be disconnected)."

I am using eeglab 2020 and Matlab R2013b. I also tried out the eeglab develop version, but the warning also occured in this version. The warning was not present in eeglab 2019.

<br>
#### Steps to Reproduce

>> eeglab
Some menus items hidden. Use Preference menu to show them all.
eeglab: options file is C:\Users\eeg_options.m
EEGLAB warning: there can be only one EEGLAB window, closing old one
Retrieving plugin versions from server...
Retreiving download statistics...
Cannot connect to the Internet to retrieve statistics for extensions
EEGLAB: adding "ICLabel" v1.2.6 (see >> help eegplugin_iclabel)
EEGLAB: adding "clean_rawdata" v2.2 (see >> help eegplugin_clean_rawdata)
EEGLAB: adding "dipfit" v3.3 (see >> help eegplugin_dipfit)
EEGLAB: adding "firfilt" v2.4 (see >> help eegplugin_firfilt)
Warning: Out of range or non-integer values truncated during conversion to character.
(Type "warning off MATLAB:nonIntegerTruncatedInConversionToChar" to suppress this warning.)
 
> In eeglab at 1154 
Could not check for the latest EEGLAB version (internet may be disconnected).  

<br>

#### Versions

|   |   |
|----| ---- | 
| OS version     | [Windows 10] |
| Matlab version | [Matlab R2013b] |
| EEGLAB version | [eeglab v2020.0] |

arnodelorme, 2020-12-02:
Yes, I think it is due to using an old version of Matlab although I would need to see where the warning message originate from. If you type

dbstop if warning

Then execution will stop when the warning message is generated. You can then send me the line number.

I will close this bug for now, but please reopen it if you have a chance to send me the information above.
Thank you for reporting the problem.

Falk8193, 2021-03-16:
I have got the same problem but I am using Matlab version R2015a and EEGLAB 2020.
Maybe you could help me as well

P.S I have just started to work with Matlab and EEGLAB

Hamdusena, 2023-01-30:
> yes, I had this problem too. I used version 14.2.b. problem is solved

arnodelorme, 2023-09-08:
Yes, I see. This is a developer feature. It would take too much effort to fix. I have added a message to clarify that issue though.

```
You have the option enabled to check for pending plugins which could be responsible
for the warning above. Please disable that option.
```

closed 2023-09-08 by arnodelorme