sccn/eeglab #223: Classify component using IClabel not working

FazlaMashrur, 2020-11-21:
### I have done ica decomposition. Not I want to classify components using IC Labels. But it's not working. 

[ Undefined function 'pop-iclabel' for input arguments of type 'struct'
Screen shot: https://www.dropbox.com/s/rxmlun6d24oaoli/Capture.PNG?dl=0
Dataset : https://www.dropbox.com/sh/gtlzjz8dw7bzsth/AAD3Wg4Lfhs9oeadnyGDZo5Ma?dl=0]

<br>
#### Steps to Reproduce

1. [*Load the data by clicking load existing dataset *]
2. [*Click tools*]
3. [*Classify components using IC Labels*]

<br>
#### Expected behavior:
[* It should give me a new window*]

<br>
#### Actual behavior:
[*Gave error saying "Undefined function 'pop-iclabel' for input arguments of type 'struct'"*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*Windows 10*] |
| Matlab version | [*2020a*] |
| EEGLAB version | [*2020_0*] |

arnodelorme, 2020-12-02:
This is because you cloned the EEGLAB repository without its submodules I think. Let me check.

arnodelorme, 2020-12-02:
Hum, this does not seem to be the case. One reason for the problem would be starting EEGLAB and then removing paths manually. Otherwise, I do not know. I would need more context. I would try to reinstall EEGLAB and see if it fixes the problem.

Closing this report for now, but please reopen if you add more information or context.

closed 2020-12-02 by arnodelorme

FazlaMashrur, 2020-12-02:
Thanks a lot. It is solved.