sccn/eeglab #226: readneurolocs function undefined

arnevdk, 2020-11-26:
### Description

I use the loadcurry plugin to load a SynAmps Neuroscan `.dat` file.
While this imports most of the EEG-data and metatada, the `loadcurry` function does not load the channel locations.
The documentation of the `readlocs` mentions it can load channel locations from Neuroscan `.dat` files.
However, for `.dat` files `readlocs` tries to call a function called ` readneurolocs`, which does not seam to be included in the current release of the EEGLAB code.

### Steps to reproduce

```
% With data.dat a Curry 6 or 7 file
readlocs('data.dat')
```

### Expected behavior
Channel locations are loaded.

### Actual behavior
```
readlocs(): 'dat' format assumed from file extension
error: 'readneurolocs' undefined near line 413 column 15
error: called from
    readlocs at line 413 column 13
```

  |  
-- | --
OS version | Fedora 32
Octave version | 5.2.0
EEGLAB version | v2020_0

arnevdk, 2020-11-26:
This mailing list thread seems to mention a version of the `readneurolocs` functions which was present at some point on the EEGLAB subversion https://sccn.ucsd.edu/pipermail/eeglablist/2012/005170.html

arnevdk, 2020-11-27:
The channel locations do get correctly loaded with loadcurry using Matlab r2020b instead of Octave

arnodelorme, 2020-12-20:
This is because you need to install the neuroscan plugin. This function is now distributed with the neuroscan plugin. I will move it back to the main EEGLAB distribution.

closed 2020-12-20 by arnodelorme