sccn/eeglab #227: Error in function pop_loadbva()at line 48

zzzznsrkl, 2020-11-30:
### Description

I'm attempting to import a bp file. The process fails with error:

EEGLAB error in function pop_loadbva() at line 48:

Quote "Channels" that does not exist

<br>
#### Steps to Reproduce
Inside eeglab
1. file>import data>from brain vision analyzer matlab file
2. click OK
3. error

<br>
#### Expected behavior:
import matlab file successfully

<br>
#### Actual behavior:
burst out an error window

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | windows 10.0.18363 |
| Matlab version | R2014a |
| EEGLAB version | i cannot find the way of searching EEGLAB version, I THINK  |

arnodelorme, 2020-12-02:
Would you mind to share the data file? Note that this function is only for Brain Vision Exchange Format data files not for Binary Brain Product files.

hy19950429, 2021-12-07:
I met the same problem, when I try to import a mat.file, the process fail with error: EEGLAB error in function pop_loadbva() at line 48:Quote "Channels" that does not exist.
Have you solved the problem, could you give me some suggestions?

nanditabhanja, 2023-09-01:
[image]

arnodelorme, 2023-09-01:
Would you mind to share a data file?

nanditabhanja, 2023-09-02:
sub1_comp.mat
<https://drive.google.com/file/d/1pXxrb4TLb776ByxKsuO3YjjekGAloJmy/view?usp=drive_web>
Please find the attached data file.

EstebanSarrias, 2023-10-16:
Hi!!

How did you resolve this?

Best regards,
Esteban

arnodelorme, 2023-10-16:
I do not think it was resolved

arnodelorme, 2024-02-10:
I asked to access the file.

cnszx123, 2024-08-19:
> I asked to access the file.

Hi, can you reproduce the issue using the file? 

I have this problem while  importing a mat file too, and I don't think it looks like a data set problem.

arnodelorme, 2024-09-09:
I have requested access to the file

https://drive.google.com/file/d/1pXxrb4TLb776ByxKsuO3YjjekGAloJmy/view?usp=drive_web

fattyuse, 2025-04-10:
Hi, can you reproduce the issue ?

I have this problem while importing a mat file too
best regards

fattyuse, 2025-04-10:
[image]