sccn/eeglab #228: Errors occur when importing .BDF file in EGGLAB

usernameseong, 2020-12-04:
### Description
[*Description of the bug or feature
Attach a link to Dropbox or Google Drive with screenshots and your data if possible.*]

I conducted an experiment with 35 channels and got a .BDF file on Actiview.  (Image 1 is the screen of experiment data on Actiview during experiment.) But this resulted in 271 channels because of some mistakes on setting. So I want extract only 35 channels from the file but there are some problems.

Image 1. Actiview screen
https://drive.google.com/file/d/1Zc4XCvbrgTUeHi3X5gxKO2kmKAlUmFn4/view?usp=sharing

<br>
#### Steps to Reproduce

1. [*First Step*]
Importing .BDF file on EGGLAB.
2. [*Second Step*]
Removing other 239 channels and saving only 35 channels on MATLAB with some codes.
3. [*and so on...*]
Importing small size dataset (from second step where I failed to remove invalid channels but size has decreased as it was captured by short period) and locating channels with '32chan.loc' file in 'sample_data' folder.

<br>
#### Expected behavior:
[*What you expected to happen*]
Now I guess 3 ways to solve this:
1. Importing only 35 channels to EGGLAB from .BDF file by removing invalid channels correctly.
2. Importing all 271 channels of huge size file and locating channels correctly.

<br>
#### Actual behavior:
[*What actually happened*]
1. [*First Step*]
Importing .BDF file on EGGLAB. - But this failed because the size of the file exceeds 12GB.
2. [*Second Step*]
Removing other 239 channels and saving only 35 channels on MATLAB with some codes - But when I checked the result dataset on EGGLAB there were still 271 channels and I've got some messages here. And when I checked this data the graph showed just parallel curves. (Image 2) I couldn't solve errors on code. (Image 3-1, Image 3-2)
3. [*and so on...*]
Importing small size dataset (from second step where I failed to remove invalid channels but size has decreased as it was captured by short period. It was for conserving the whole time and removing the number of channels but it went nonsense.) and locating channels with '32chan.loc' file in 'sample_data' folder. - Unfortunately I got a message that says "The number of data channels (32) not including fiducials does not correspond to the initial number of channels (271), so for consistency purposes (...)" (Image 5) and I couldn't solve this. Then I tried with BESA file but the result only showed 18 channels. 

Image 2. Small sized dataset plot graph
https://drive.google.com/file/d/1y6IhOT1zWCt2DHWZYwVmP6uUPe9ektXG/view?usp=sharing

Image 3-1. Options when removing channels. Here I only selected the range of channels 
https://drive.google.com/file/d/1CFF6ZU_NDZJT_1Eai2TTY4aM7Lhju9Lz/view?usp=sharing

Image 3-2. Error messages I got during running the code.
https://drive.google.com/file/d/1S3qD_40o4RJUzn08Y2YENNNmj9FySIpN/view?usp=sharing

Image 4. Edit channel info screen
https://drive.google.com/file/d/1JEUZNaPajNwZ48jAAD-bDNRb3msTxMw9/view?usp=sharing

Image 5. The number of data channel message
https://drive.google.com/file/d/17UzT6ULE9qTuB0cgUx3NlTMT23lO2n8L/view?usp=sharing

Source 1. Code for removing invalid channels
https://drive.google.com/file/d/1HZE7vWC78K35hgBPdG6DJbrJVqsO_nZR/view?usp=sharing

Source 2. Code for locating channels while removing invalid channels - But this didn't work so I used the file on EGGLAB like chan32.loc or BESA.
https://drive.google.com/file/d/1kCzuCbVc5l59JSXW3ExmlETxoEQ7ZrRY/view?usp=sharing

Source 3. Excel file for locating channels while removing invalid channels
https://drive.google.com/file/d/1xd3S7K3OYziWIXIOASLx_T2KTeDNO82K/view?usp=sharing

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [Windows 10] |
| Matlab version | [2020b] |
| EEGLAB version | [2019 released] |

arnodelorme, 2020-12-04:
Would you mind to share the BDF file?

usernameseong, 2020-12-04:
> Would you mind to share the BDF file?
I'm sorry I cannot share the file now because of some personal information issues. Could you suggest other ways for solving this? Thanks.

closed 2020-12-07 by usernameseong