sccn/eeglab #229: Error when using xdfimport plugin to open multistreams xdf file

Mermaidlegs, 2020-12-08:
### Description

When I was using xdfimport plugin to import the xdf data, after I put the stream name, stream type and exclude marker streams, the error shows that "Bracing is not supported for the variables in this type". 

<br>
#### Steps to Reproduce

Please see the google doc for more details.
https://docs.google.com/document/d/1cGrUdacgLJMj5mNk-oY6EQseFqGVjy3YaPyelpG_Uts/edit?usp=sharing

<br>
#### Expected behavior:
Open the xdf file

<br>
#### Actual behavior:
pause at the line 606 when calling load_xdf function, says: 'brace indexing is not supported for variables of this type'

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [Mac OS Catalina 10.15.3] |
| Matlab version | [R2020b] |
| EEGLAB version | [2020] |
| xdfimport version | [v1.16]

arnodelorme, 2020-12-20:
I have requested access to the file. Sorry it took so long. What software did you use to collect the file?

Mermaidlegs, 2020-12-20:
> I have requested access to the file. Sorry it took so long. What software did you use to collect the file?

I have added your email account as editor to the file. Thank you so much. We used g.tec: https://www.gtec.at

arnodelorme, 2020-12-20:
I saw the error in the document you shared. We would need access to the raw XDF file though. Thank you.

arnodelorme, 2021-01-15:
I can see the document now, but we would need access to the raw .xdf data file.

Mermaidlegs, 2021-01-15:
Attachment available until 14. Feb 2021
Dear Dr. Arnaud,
Please find the pdf file attached. 
Thank you for the help.
Click to Download
 <https://www.icloud.com/attachment/?u=https%3A%2F%2Fcvws.icloud-content.com%2FB%2FAQPQszeXJXrt9rIrxRfLarlfR5zfAU-kmCRjNZZ92xDbGPW1qgGT5cHT%2F%24%7Bf%7D%3Fo%3DAi_LNwXH0AkXehtI5wIEKI5V2HxGEddJCccquTWeRS3D%26v%3D1%26x%3D3%26a%3DCAog-q0V0l7vUPjAmf_FlJOC-ZBOZg7gbJmGgHq9MXFlQU0SehDf-tK38C4Y34rOi_ouIgEAKgkC6AMA_y-PEv9SBF9HnN9aBJPlwdNqJ4g8KOzHI-IiGVQ3m8ZFJMYROKIqk21s4rs0wtX78N3n5ev7L4dKy3InfGWk4x07J6JU8U6o-zzu2bniDaqt4TBJXgXhm9DhnjrTjLsVzfiW%26e%3D1613321438%26fl%3D%26r%3DE0542332-656F-4F01-919A-468E3612FF54-1%26k%3D%24%7Buk%7D%26ckc%3Dcom.apple.largeattachment%26ckz%3D616EDA91-F107-401E-8A12-777EB4764459%26p%3D42%26s%3DZkEqU2JPoUkPQ_ZKnS7Cpm50Ox0&uk=SqWAig70pJTJSaKC_vgWjQ&f=ANB87.xdf&sz=739091467>ANB87.xdf
739,1 MB
Best wishes,
Lin

arnodelorme, 2021-01-16:
We have issued version 1.18 of the plugin, which contains a fix. Let us know if it works for you. Most likely g.tek does not save the data properly, but I am not going to argue.

closed 2021-01-16 by arnodelorme