sccn/eeglab #231: pop_eegplot() note: Baseline subtraction disabled to speed up display

baharkk20, 2020-12-11:
### EEGPlot

[*Couldn't plot the data
Attach a link to Dropbox or Google Drive with screenshots and your data if possible.*]
[https://vumc.box.com/s/27h8ux6goysxqbl6nd3e9gcv7pw0s6sf](url)
1. I imported the data that is collected at EGI net station
2. Marked the R wave 
3. Plot -> channel data (scroll)

 I couldn't open the plot. When I try this is seen : 'pop_eegplot() note: Baseline subtraction disabled to speed up display' with this error: 

[image]

EEGLAB error in function eegplot() at line 1149:
Matrix dimensions must agree.

| OS version     | [*Windows 7 Professional Version 6.19 (Build 7601:Service Pack 1)*] |
| Matlab version | [*R2020a-academic use*] |
| EEGLAB version | [*EEGLABv2019.1*] |

Thank you so much for your time! 
Bahar

closed 2020-12-17 by baharkk20

arnodelorme, 2020-12-17:
What was the problem?

baharkk20, 2020-12-18:
Hi Arnaud,

The data was collected at net station and marked for the heart beats but I couldn't plot when I load the .mff file. And then I figure out that when I load the .set file that is marked for heart beats, I could plot.
I had a new problem though and may be you could help with that or have an idea. The heartbeats marked through biopac and there is a shift at the marked beats (it is at -100ms or -150 ms) and I couldn't figure out what causes this problem.

Thank you so much for your time!

Kind Regards,
Bahar

What was the problem?

[ WARNING : This email came from an external source. Please treat this message with additional caution.]