sccn/eeglab #232: Error occured in function datenum() at line 188

MartinS071, 2020-12-11:
Hi,

When I want to import a mmf file, I get this error DATENUM failed (Error occured in function datenum() at line 188).

This occurs only on one computer, it works on my other computer. All my MATLAB scripts and EEGlab plungins are in my Dropbox. I wonder if it may cause problem.

Regards,

Martin

arnodelorme, 2020-12-20:
I do not think dropbox is the problem. Can you share the file?

MartinS071, 2020-12-20:
Hi Arnaud,

I found out the issue. It is likely related to MATLAB version. MATLAB 2019b runs on the computer without problem. On the computer with this problem, MATLAB 2020a is installed. To solve the problem, I edited line 45 of pop_mffimport.m:

if datenum(matVer(1,1).Date) < 735595 % Martin change datenum(matVer.Date) pour datenum(matVer(1,1).Date)
    error('This version of Matlab is too old. Use version 2014a or later');
end

I did this as matVer.Data size was (1,2).  I presume it is related to MATLAB version as I did not get the error when processing data using MATLAB 2019b.

Regards,

Martin

[Externe UL*]

*ATTENTION : L’émetteur de ce courriel est externe à l’Université Laval.
Évitez de cliquer sur un hyperlien, d’ouvrir une pièce jointe ou de transmettre des informations si vous ne connaissez pas l’expéditeur du courriel. En cas de doute, contactez l’équipe de soutien informatique de votre unité ou hameconnage@ulaval.ca<mailto:hameconnage@ulaval.ca>.

arnodelorme, 2020-12-20:
2019b is not old. If you share the file with me, I will add it to our list of files to test on, on Windows and Mac. But if you prefer not to do that, this is fine. I am closing the bug report right now.

closed 2020-12-20 by arnodelorme

MartinS071, 2020-12-20:
I will be happy to share the file. This size is ~545 Mo. How can I share the file?

[Externe UL*]

arnodelorme, 2020-12-20:
Put it below (drag and drop)

https://drive.google.com/drive/folders/1yoJ19QN2uJOzyDs4eiD9T7dI7qDYs2Th?usp=sharing

MartinS071, 2020-12-21:
Arnaud,

I have downloaded the MFF file.

As I told you, matVer size was 1 x 2 thus it causes the error. This is related to a “Toolbox” called GLMLAB: fitting generalised linear models. It is not a MATLAB toolbox but a “toolbox” from MATLAB central. It seems that ver(‘MATLAB’) returns this toolbox while it ignores all other toolboxes. This might be due to a simple comma… see screen capture.

Thanks and happy holidays,

Martin

[Externe UL*]

https://drive.google.com/drive/folders/1yoJ19QN2uJOzyDs4eiD9T7dI7qDYs2Th?usp=sharing