sccn/eeglab #233: 'Select epochs or events', amending events' time

RenzoLanfranco, 2020-12-17:
### Description

This isn't an issue but a suggestion for improvement. When you select an epoch or event (pop_selectevent()) in EEGLAB's GUI, it would be extremely helpful if you added an option to amend events' time. More specifically, to select one event that is found in all epochs and change their temporal location. Currently, EEGLAB only allows renaming event types. One example: if you have an event '11' that corresponds to a keypress in every epoch, it would be very helpful to have the option of selecting all those epochs and changing their temporal locations (variable across epochs) to one temporal location (e.g. 0 ms).

<br>
#### Steps to Reproduce

1. Edit -> Select epochs or events ->
2.  (in new window) Selecting type of event

<br>
#### Expected behavior:
To have another box allowing to edit the selected event's temporal location. 
For example, to be able to select an event that is printed in different temporal locations across trials and relocate it at time zero in every trial.

<br>
#### Actual behavior:
Such an option isn't currently available.

<br>
#### Versions
EEGLAB v2020.0

arnodelorme, 2020-12-20:
Could you describe a little bit more what you would want to do with these modified events?

amisepa, 2021-02-25:
@arnodelorme here is what he wanted to do and the solution I had sent him via EEGLABLIST (in case that can be helpful):

Renzo: Based on these event codes, I need to create new event codes for every
trial and print new codes at time zero that can comprise the previous event
codes. For example, if event code at time zero can be 1, 2 or 3, and
response code printed later can also be 4, 5 or 6., I now need to have
event codes at time zeroes that account for all those combinations, e.g.
14, 15, 16, 24, 25, 26, 34, 35, and 36.

Test data:
[Attachments-RE [Eeglablist] Creating new event codes based on previous combination of event codes in each trial.zip](https://github.com/sccn/eeglab/files/6039657/Attachments-RE.Eeglablist.Creating.new.event.codes.based.on.previous.combination.of.event.codes.in.each.trial.zip)

Solution I found:
[amending_event_times.zip](https://github.com/sccn/eeglab/files/6039682/amending_event_times.zip)

 attached.

arnodelorme, 2022-05-18:
Yes, this would be a good topic for a plugin. Happy to test. Closing this bug report for now.

closed 2022-05-18 by arnodelorme