sccn/eeglab #242: Issue with epoch end (sec) when splitting continuous recording using eeg_regepochs

Jazelle759, 2021-01-11:
### Description

I am not too sure if this is an issue or I misunderstood how to use eeg_regepochs.

After applying average reference to my continuous data, the epoch start (sec) is 0 and epoch end (sec) is 432.660.
I wanted to split the data into 10 seconds epochs so I tried using eeg_regepochs to split it.
I was hoping it will be divided into 43 epochs with each epoch start to be 0 and end to be 10. 
It did get divided into 43 epochs with start of 0 but the epoch end was 0.996 instead of anywhere near 10 secs which seemed odd.

Screenshot links
[before_epoch_screenshot](https://drive.google.com/file/d/1nmuMdgwoUrVb0mZh8ump3W0SWSz6-RNb/view?usp=sharing)
[after_epoch_screenshot](https://drive.google.com/file/d/1qyfPZEeAaKDw6ib0DpBDKZ4xZU5Zr34K/view?usp=sharing)

I've uploaded the dataset I tried using eeg_regepochs on. Steps has been applied until average reference step using the code below before saving.
[Example Dataset](https://drive.google.com/file/d/1c9lX9aY-G7cUeoWu9-FykMBngfkjafDa/view?usp=sharing)

<br>
#### Steps to Reproduce

1. Downsample data to 250
`EEG = pop_resample( EEG, 250);`

2. Bandpass filter using IIR Butterworth filter in ERPLAB (1-30)
`EEG  = pop_basicfilter( EEG,  1:128 , 'Boundary', 'boundary', 'Cutoff', [ 1 30], 'Design', 'butter', 'Filter', 'bandpass', 'Order', 2 );`

3. Add channel locations
`EEG = pop_chanedit(EEG, 'load',{'\GSN-HydroCel-129.sfp' 'filetype' 'autodetect'},'changefield',{132 'datachan' 0},'setref',{'1:130' 'Cz'});`

4. Apply average reference
`EEG = pop_reref( EEG, [],'refloc',struct('labels',{'Cz'},'Y',{0},'X',{5.4492e-16},'Z',{8.8992},'sph_theta',{0},'sph_phi',{90},'sph_radius',{8.8992},'theta',{0},'radius',{0},'type',{''},'ref',{''},'urchan',{132},'datachan',{0}));`

5. Create epochs of 10 seconds each
`EEG = eeg_regepochs(EEG, 'recurrence', 10.0, 'rmbase', NaN);`

<br>
#### Expected behavior:
Epoch end (sec) to be 10

<br>
#### Actual behavior:
Epoch end (sec) 0.996

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | Windows Server 2016 Standard |
| Matlab version | 9.5.0.1033004/R2018b |
| EEGLAB version | v2020.0 |

ronjafa, 2021-08-13:
Hello

I had the same problem and through trial and error came to the solution. Namely, I think that you also have to define ‚limits'. This code worked for me to segment my data into 4 second epochs non overlapping.

EEG= eeg_regepochs (EEG, 'recurrence', 4, 'limits', [0 4])

gdemirezen, 2022-01-07:
Yes, calling the function with defining limits explicitly is a work-around.

I see the problem is related to the usage of nargin in eeg_regepochs.m. 
For example, the call eeg_regepochs(EEG, 'recurrence', 2) results in a local nargin value of 3.
Lines 106-108:
  if nargin < 3
    g.limits = [0 g.recurrence];
  end
Since nargin=3, g.limits is not set as the default value of [0 recurrence_interval] as written in the function description but remains at [0 1], which was the default value input to finputcheck() at line 95.
Then at line 130, eplength is calculated as 1 since g.limits = [0 1] though it needed to be 2 in this case.

A control structure around lines 106-108 as in lines 87-94 can be used as a solution if ordered values without keys are to be input as well.

Note: I am using eeglab2021.1 and MATLAB 2020a.