sccn/eeglab #243: Channel location vhdr file

arnodelorme, 2021-01-14:
Hello eeglab gurus,

I am processing EEG data by loading .vhdr files from BrainVision amplifier.
When I loaded the data into EEGLAB, I could not edit channel locations by
"Edit - Channel Locations", I would get an error message saying:

"*Operands to the II and && operators must be convertible to logical scalar
values.*

*Error in pop_chanedit (line 163)*
*if isempty(chans) || ~ishandle(chans)*"

The .vhdr files I used contain the standard channel names, and I thought
there would be a way to get the default channel locations by EEGLAB. But I
don't know how to do that. Could you please advise on how to resolve this
problem so I can get the channel locations (to make topographical plots)?

Thank you very much,

bokiwang, 2021-01-14:
The data, header file, and VMRK files are included in the zip file. Thank you for your help!

[Channel Location files.zip](https://github.com/sccn/eeglab/files/5816374/Channel.Location.files.zip)

arnodelorme, 2021-01-14:
Hi Boki, I have downloaded and imported your data both in the dev version of EEGLAB and with EEGLAB 2020.0 using the BVA-IO plugin version v1.7.

1. Import menu "File > Import data > Using EEGLAB function... > From BVA .vhdr ...", then press OK with the defaults.
2. Then edit channel location Edit > Channel locations, press OK. Everything was fine.

Maybe you are using an old version of EEGLAB or an old version of the plugin.
Best wishes

closed 2021-01-14 by arnodelorme

bokiwang, 2021-01-15:
Yes you're right! once I updated to the latest EEGLAB and bva-io plug in,
everything worked. Thank you Arnaud!

arnodelorme, 2021-01-15:
No prob