sccn/eeglab #246: Error in Physical maximum and minimum values for EDF in writeeeg

SparshJ, 2021-01-17:
Hey!

While using the writeeeg function on a data matrix ( channels X samples), the resultant EDF shows different physical maximum and minimum values for different channels even though no such info is passed to writeeeg. This error occurs only for some subjects, whereas for other subjects the physical maximum and minimum values are identical across the channels. Please help!

Thanks!!

arnodelorme, 2021-02-01:
Would you mind to provide an example demonstrating the problem? We will look into it.

SparshJ, 2021-02-01:
Sure!
I've attached the screenshot below:
[image]
As you can see, the physical max and min values are not consistent across the channels. As a result, its not possible to create bipolar montages by subtracting two channels. I think that the swrite function called by pop_biosig's writeeeg calculates these physical max and min values, and that's where this discrepancy is occurring.

Thanks!

arnodelorme, 2021-02-01:
Would you mind sharing the file you are trying to export. It is unclear what the spreadsheet represents. @alois.schloegl@gmail.com what do you think?

SparshJ, 2021-02-01:
Hey! Here's the drive link to the file:
https://drive.google.com/file/d/1WYBXzYVLsp234j4DZwvo1c44UecmH-uR/view?usp=drivesdk

Teuniz, 2021-05-04:
As a workaround, use the tool "Unify resolution", it's in the Tools menu of EDFbrowser.  
  
https://www.teuniz.net/edfbrowser/EDFbrowser%20manual.html#Unify_resolution

arnodelorme, 2021-07-16:
@amisepa you are the EDF expert. Would you mind checking if we can close this report?

amisepa, 2021-07-23:
Sorry, I don't know much on this. @Teuniz In what situation is the physical value not homogenous? Impedance during recording? Glitchy electrode? Can you share how you unify the resolution in EDFBrowser so we could add a similar feature in EEGLAB?
Thanks,
Cedric

Teuniz, 2021-07-25:
It appears that the writeeeg function uses the actual peak values found in the signal's samples for pys max and phys min.  
This is not as intended by the EDF format.  
Pys max and phys min should be the clipping levels of the ADC input and they should be the same for all channels.  
If this information is not available, for example when converting from another format, use a safe value for all EEG channels  
which is, for example, +3000 uV and -3000 uV.  
For example, Nihon Kohden uses +3200 uV and -3200 uV for all EEG channels (which are the actual clipping levels  
of their input amplifiers & ADC.

amisepa, 2021-07-26:
@Teuniz  Thank you very much for pointing it out. 

@arnodelorme I'll let you decide whether you make this adjustment in writeeeg.

arnodelorme, 2021-07-26:
This is a reasonable assumption, @cll008 would you mind looking at it. Also, test files https://github.com/amisepa/import_EDF

cll008, 2021-07-27:
@amisepa Thanks for your input Cedric. @arnodelorme asked me to unify based on the most extreme values in the recording. I implemented it, but maybe we should be cautious about EXG or AUX channels that could have different enough scales to cause problems with the rest of the EEG channels?

cll008, 2021-07-27:
Is the ADC range somewhere in the metadata? Looks like `writeeeg.m` uses a digital range of [-32768 32767] (min is off by 1 compared to FAQ below?) so maybe we can use that for the physical range too...

Excerpt from [EDF FAQ](https://www.edfplus.info/specs/edffaq.html)

> Q4. Are the "digital minimum" and "digital maximum" values hints or strict limits?
> The specs say "The digital minimum and maximum of each signal should specify the extreme values that can occur in the data records." Note the word "can". It is not necessary that these values actually DO occur. So take safe values that you know the signal will not exceed, for instance the range of the ADC. Note that "The physical (usually also physiological) minimum and maximum of this signal should correspond to these digital extremes". This correspondence is necessary for assessing gain and offset of the signal.
Q5. Why not always use -32767 for "digital minimum" and +32767 for "digital maximum"?
Export. It is formally correct EDF as long as the purpose (specification of offset and amplification of the signal) is met with sufficient accuracy.

Teuniz, 2021-07-28:
EXG and AUX channels usually have different scales/ranges compared to EEG channels.  
If available, use the values in the header of the original EDF file that has been used for the input.

cll008, 2021-07-28:
@arnodelorme we can store PhysMax and PhysMin in EEG.etc https://github.com/sccn/eeglab/pull/360/commits/33c425cee5fa6e81732fa9806c12b5d073f8c21b and pass this as an argument to pop_writeeeg and as an option to writeeeg. Thoughts?

amisepa, 2021-07-29:
Funny, I think I just run into this problem randomly trying to export some preprocessed data as .bdf or .edf:

[image]

I think we should hardcode the physMax and PhysMin to the values suggested by Teuniz above. 
Maybe we can use detection for AUX channels only?

arnodelorme, 2021-08-16:
@cll008 
1. Test if export is a problem (import, export, reimport) -> this should mess up amplitude
2. Implement the -32768 to 32768 max limits

amisepa, 2021-09-01:
@cll008 I still get this error when importing a .bdf file with biosig toolbox, and try to export it as .bdf
[image]
Looks like the physical min/max is not allocated to each channel maybe?

cll008, 2021-09-02:
Thanks for the input and the summary from the other thread Cedric. The PhysMax and Min are in, but I'm going to check the rest of your test files to see if +-5353 is a more common range than +-3200. PhysDimCode seems to code for the unit, I guess `writeeeg` never cared about this but it's pretty important. I also wonder if we should change the data length so that the sampling rate doesn't get fudged...

amisepa, 2021-09-02:
Just to clarify:
My error comes from exporting a file (BDF or EDF that I preprocessed in EEGLAB) as .BDF or .EDF, to allow someone to visualize it easily in EDFBrowser. So I think my error is from the exporting functions. But it indicates that there might be something going on with the physMax/Min information somewhere (specifically, the pshyMax/Min is expected to be defined for each channel, which is not the case). That's why I shared it here. I don't know if my test files are the best to base this on as I don't even know what system recorded them. 
From that other thread, looks like it's a common issue that might be specific to exporting. And that there might not be a clear solution. 

Is it possible to define PhysMax/Min for each channel in writeeg or pop_writeeg? Or do you need to expand dataduration to do that?
See the other thread regarding data duration if you are going to change it, as it doesn't sound very straightforward.

cll008, 2021-09-02:
Yes currently in `writeeeg` PhysMax and Min can be defined. I've been testing the changes downstream in EDFBrowser too. @arnodelorme decided we want to hardcode this to a reasonable value (unlike the previous scheme which leads to problems by using each signals' max). Really the best is to keep the header info if available but I guess we don't want to add another field to the EEG struct.

cll008, 2021-09-07:
Turns out the test files have inconsistent `PhysMax` (5353, 5000, 8092, 8719, etc.) so it seems using the max of all channels is a good way to go (barring storing header info on read). I also found a `PhysDimCode` of 4256+19 codes for uV so I added that to prevent our warning message on import. (`PhysDim` of uV is already coded for so shouldn't actually cause any issues)

amisepa, 2021-09-10:
Sounds good. FYI: I don't know how these files were recorded and with what system, so I don't know if we should base this decision on them (although helpful). Maybe you can double-check with some more standard files (e.g., biosemi file attached)

[6e8691ffd0_eyesclosed.zip](https://github.com/sccn/eeglab/files/7141235/6e8691ffd0_eyesclosed.zip)

Cedric

PR #381 (2021-09-20): writeeeg update PhysMax, PhysMin, PhysDimCode
fix #246. Change PhysMax and PhysMin scheme, add PhysDimCode
closed 2021-09-20 by arnodelorme