sccn/eeglab #247: Compatibility of EEGLAB with "Matlab Online"?

olafdimigen, 2021-01-19:
### Question: Are there any experiences (or successful attempts) with running EEGLAB on _Matlab Online_ (using the GUI)?

Issue: EEGLAB does not seem to run on "Matlab Online* (http://matlab.mathworks.com), due to GUI issues

#### Steps to Reproduce
1. Matlab Online is a tool to run MATLAB code in a browser environment that emulates Matlab
2. Compatibility with Matlab Online would be highly useful to run workshops remotely that require that all participants run EEGLAB in a standardized software/hardware environment 
3. EEGLAB starts in Matlab Online, but then throws various errors related to GUI callbacks (happy to document in detail)

Tested with: **EEGLAB 2019_1** and the **EEGLAB dev. version** (version Jan-19, 2021) using Win10 & Firefox

arnodelorme, 2021-01-20:
Yes, I am able to reproduce some errors.
If I put breakpoints, sometimes they disappear. It seems to be a complex Matlab bug. I will report it to Mathworks.

arnodelorme, 2021-01-20:
[image]

closed 2021-01-20 by arnodelorme