sccn/eeglab #251: EEGLAB error in function pop_loadbv() at line 87

NicoBiagi, 2021-01-21:
Hi, I tried to load a .dat file using the 'From Brain Vis. Rec. .vhdr or .adhr file' option from the bv-IO1.7 plugin.
I left both the 'Interval samples' and the 'Channels' blank because I was happy with the default values.

However, I get an error message saying: 'Output argument "CONF" 9and maybe others) not assigned during call to "readbvconf".

Could you please help me fix this error?

Many thanks

arnodelorme, 2021-01-21:
Dear Nico, Would you mind to share your data files?

NicoBiagi, 2021-01-21:
Hi,

Thank you for your prompt reply.

I had to attach the data in .zip format because GitHub didn't let me add it as a .dat file

Best,
Nico
[Example_data.dat.zip](https://github.com/sccn/eeglab/files/5851199/Example_data.dat.zip)

arnodelorme, 2021-01-21:
Thank you. We would need the associated .vhdr and the .vmrk files as well to be able to run some tests.

NicoBiagi, 2021-01-21:
I don't have the .vhdr file, just the .vmrk file
[Example_data.vmrk.zip](https://github.com/sccn/eeglab/files/5851369/Example_data.vmrk.zip)

arnodelorme, 2021-01-21:
If you do not have these files, then the data cannot be imported. The files always come in 3 (a .dat, a .vhdr, and a .vmrk file). Best

arnodelorme, 2021-01-21:
The .vhdr is the header file and it is the most important. the .vmrk is the marker file (events) and it is optional.

closed 2021-01-21 by arnodelorme

NicoBiagi, 2021-01-21:
here is the .vhdr file relative to the example dataset I've sent you
[Example_data.vhdr.zip](https://github.com/sccn/eeglab/files/5851473/Example_data.vhdr.zip)

arnodelorme, 2021-01-21:
It is not a good idea to rename files because they refer to each other. When the files are named properly, they can be imported.

arnodelorme, 2021-01-21:
[Archive.zip](https://github.com/sccn/eeglab/files/5851508/Archive.zip)