sccn/eeglab #253: Error with plotting cluster ERPs - STUDY

denizdor, 2021-01-29:
Hello Arnaud,

Thanks for helping me with my previous question regarding the problem I had with initiating EEGlab2021 on Matlab2016a. 
Here I wanted to report a different problem, and I believe this was previously addressed but I can't seem to have it work even after following the corrections you recommended here: https://github.com/sccn/eeglab/issues/204#issue-690142677

I tried this on EEGlab 2019/2020/2021, and Matlab 2016a, and now 2020a (just upgraded to a newer version since I thought maybe it was again Matlab 2016a causing problems). All have resulted in the same error.

To be more specific:

I have a small dataset with 8 subjects that I am testing out different study functions and LIMO. Each dataset belong to an independent participant and consists of 2 different conditions within the dataset and each has been decomposed by ICA as a single dataset. Then I have a second STUDY variable 'group' which has 2 levels (Group 1= 4 subjects, and group 2=4 subjects). 

I am able to cluster components (rv at 15%) and plot scalp maps and dipoles with no issue. However similar to the previously reported issue, when I try to plot ERPs or Plot ERP image on clusters I receive the following error:

[image]

I can share my data too if it would be helpful. 
Thanks a lot!
Deniz

| OS version - Big Sur
| Matlab version | replicated the error in both 2016a and 2020a 
| EEGLAB version | replicated the error in both 2019.1, 2020 and 2021.

arnodelorme, 2021-01-29:
Hi Deniz, you need to make sure that you have different sessions for your subjects because if you just have one session, EEGLAB assumes that you need one ICA decomposition for all the data of that subject. Even so, if you have different ICA decomposition for the same subjects, the type of analysis you can do is limited. Let us know if that helps.

closed 2021-01-29 by arnodelorme

denizdor, 2021-01-29:
Thank you Arnaud!
So would you recommend that I split the datasets into 2 separate datasets, creating one for each condition and then run the ICA on each subsets, or run the ICA first and then split the datasets into 2 conditions and then create the study. I think I had read in the tutorial that the latter is better (running one ICA per subject)? Both conditions are part of the same task and recorded during the same session for each subject.

I was able to analyze the ERP channel data (not the components) with LIMO in a repeated measure ANOVA with group as the between-subject factor, but I also wanted to identify the components/dipoles contributing to the significant ERPs that would correspond to my LIMO results.  I will probably write to the mail list to ask for further guidance.

Thanks again for your prompt replies, I really appreciate it!
Best,
Deniz