sccn/eeglab #254: Some XDF files channels lost and frequency changed after imported

Mermaidlegs, 2021-02-03:
### There are two problems I faced when using xdf import plugin 1.18

<br>
#### Steps to Reproduce

1. Import the xdf files by using xdf file import function
2. Some of the datasets are opened with 256 hz and 16 channels (right sampling rate and right numbers of the channels)
3. However, some of the datasets are opened with 1000 hz and 1 channel only
notes: we are sure that every dataset was recorded with the same paramteres and the same sampling rate

Please see the drive link for more details.

<br>
#### Expected behaviour:
all the datasets would be opened with the right sampling rate and right numbers of channels.

|   |   |
|----| ---- | 
| OS version     | [10.15.3] |
| Matlab version | [R2020b] |
| EEGLAB version | [2020.0] |

Thank you so much for the help and support.

Mermaidlegs, 2021-02-03:
https://drive.google.com/drive/folders/1IALV3LAI1jUOM3pnatZcNwSwXnH6w6cC?usp=sharing

The link is here.

Thank you in advance.

cll008, 2021-05-10:
I'll close this for now but please reopen if necessary. Thanks.

closed 2021-05-10 by cll008