sccn/eeglab #255: Issue loading EDF/EDF+ files

SamehHanna, 2021-02-14:
### Description
I am new to the EEGLAB (actually the whole EEG research). I am trying to load the dataset from https://physionet.org/content/eegmmidb/1.0.0/
The data is in the EDF+ format. While the channels are all read (64), it reads all as **one** epoch and **one** event. Also, no channel locations information. From what I understood it should contain the different events mentioned in the upper link. Could you explain if I did something wrong? Or missed any needed step to read the read correctly? 
Sorry in advance if it was not an issue but misunderstanding from my side.
<br>
#### Steps to Reproduce
1. File> Import data> Using EEGLAB functions and plugins> from EDF/EDF+/GDF files ==> then select S001 folder then S001R01
2. I use the default in the next screen (extract even & import annotations checkboxes on only) ==> give any name
<br>
#### Versions
|   |   |
|----| ---- | 
| OS version     | [*Windows 10 Pro*] |
| Matlab version | [*(R2020b) Update 4*] |
| EEGLAB version | [*V2021.0*] |

arnodelorme, 2021-02-15:
@amisepa this is a good problem to look into if you are interested

SamehHanna, 2021-02-18:
Thank you for your help in advance @arnodelorme @amisepa. 
In the meanwhile, Is there any work-around you could advise to read the files?

amisepa, 2021-02-18:
Hi,

I will look into it asap.
In the meantime, you could use EDF viewer to check that the information you are missing is actually in the files: 
https://www.teuniz.net/edfbrowser/

Cedric

SamehHanna, 2021-02-19:
Hi Cerdic, @amisepa 
I tried to load using EDF viewer, here is what I found.
first when I tried to open the file, it gave an error
[image]

then when I used the Tools> Header Editor and selected the file (I just saved it without any changes in default values)
and I can now open it 
[image]

Hope this can help you a little in your debugging.

Teuniz, 2021-02-19:
The file https://physionet.org/files/eegmmidb/1.0.0/S001/S001R01.edf contains only one annotation:

[image]

The file https://physionet.org/files/eegmmidb/1.0.0/S001/S001R08.edf contains 30 annotations:

[image]

I guess you should ask the authors of the article why some file(s) have only one annotation.

Regards,
Teunis

amisepa, 2021-02-19:
Hi,

I have been trying the same thing with EDFBrowser, and I also think the problem comes from the files (which might not be a problem actually, we just need to know why these trials have only 1 event).

I opened the .event files with a text editor and you can easily see the difference between:
S001R01 and S001R02 that have only one event: 
[image]

and S001R03-R14 that have many:
[image]

It appears to be the same for other subjects, so my first guess would be that R01 and R02 might be pilot runs to train subject to do the task, and 3-14 the real runs. But the authors should have a description file with their dataset to indicate that this is the case, which I am not seeing. 

If you could contact the authors to confirm this, that would be great.

It seems to work in EEGLAB when you load runs 3-14, despite the warning messages in Matlab. Here is a screenshot of Run11 for example:
[image]

Cedric

amisepa, 2021-02-19:
I am going to look into why the events are all T0 instead of T0, T1, T2 though

amisepa, 2021-02-19:
Actually, they did provide some information about events, and they had to develop a separate annotation file for their software: 
``The data are provided here in EDF+ format (containing 64 EEG signals, each sampled at 160 samples per second, and an annotation channel). For use with PhysioToolkit software, rdedfann generated a separate PhysioBank-compatible annotation file (with the suffix .event) for each recording. The .event files and the annotation channels in the corresponding .edf files contain identical data.`

I am working on a solution to get the correct event names in EEGLAB for this type of file.

SamehHanna, 2021-02-19:
Ok Great thank you for all you help. @Teuniz thanks for the clarification, I missed that (as I said I am new to the field). @amisepa Thanks for your continuous help. Please keep me updated regarding this.

amisepa, 2021-02-23:
Here is a potential fix. I attempt 2 methods depending on whether you have Matlab 2020b with Signal processing toolbox, or Matlab <2020b and no Signal processing toolbox. 

Please try:
1) download this attached pop_importedfevents.m function
2) add matlab path to it
3) import edf data with biosig toolbox as you did before with GUI or command line: 
EEG = pop_biosig('C:\Users\IONSLAB\Desktop\eegplugin_edf_plus\test_data\S001R03.edf');
4) import corrected events with the attached function:
EEG.event = pop_importedfevents();          %select file manually with pop-up
EEG.event = pop_importedfevents(fullfile('filePath', fileName.edf'));
EEG = eeg_checkset(EEG);

It works whether you check the "Import events" and "import annotations (EDF+)" boxes or not during data import, so you don't have to worry about that.
I compared with markers names and timestamps from EDFbrowser and it looks correct.
Please run more tests and let me know if everything looks good.

@arnodelorme I believe we can either:
 1) adapt line 1055 in sopen function:
HDR = edfannot2evt(HDR);
2) call my pop_importedfevents() function in pop_biosig as at line 189-203

Cedric

SamehHanna, 2021-02-23:
@amisepa Thank you Cedric, On the other hand, Could you check the file/link in your comment? Because for some reason I can't see the file/link anywhere in your comment even after I searched the whole repository for "pop_importedfevents.m" and couldn't find it.

amisepa, 2021-02-23:
@SamehHanna Sorry i forgot to provide the link to the file. Here it is: https://drive.google.com/file/d/1FQlS4kO3qG8uAhXZFH9juGqKR69irvpF/view?usp=sharing
Let me know if that doesn't work.

SamehHanna, 2021-02-23:
Great Thank you Cedric @amisepa. I will let you know once I try it.

SamehHanna, 2021-02-25:
Thank you Cedric @amisepa I managed to read the file and check the events following your instructions. here is a screenshot of it 
[image]

amisepa, 2021-02-25:
Hi,

Great. The changes are now merged with eeglab, so you can pull eeglab and use the usual Biosig import steps, just make sure you check the new box named 'Import using alternative BIOSIG method (mexSLOAD)'

Cedric

SamehHanna, 2021-03-01:
Sorry for the late reply.
Great it is Working now. Thank you all for all your help and support I am closing with this comment now. 
@amisepa @Teuniz @arnodelorme

closed 2021-03-01 by SamehHanna

SionHalim, 2021-07-23:
@amisepa can u reupload the file? i have open your google drive link but it says that "Sorry, the file you have requested does not exist." ... its from your shared link https://drive.google.com/file/d/1FQlS4kO3qG8uAhXZFH9juGqKR69irvpF/view?usp=sharing

amisepa, 2021-07-24:
Here is the updated version: https://github.com/amisepa/import_EDF

Lucreziadimarino, 2024-10-26:
> CIAO,
> 
> Ottimo. Le modifiche sono ora unite a eeglab, quindi puoi estrarre eeglab e usare i soliti passaggi di importazione Biosig, assicurati solo di spuntare la nuova casella denominata 'Importa usando un metodo BIOSIG alternativo (mexSLOAD)'
> 
> Cedrico

Hi @amisepa, i have some problems with my .edf file... eeglab can't extract events from my files and i don't know how to solve. Could you help me?

amisepa, 2024-10-28:
Hi @Lucreziadimarino 

this is a close issue. Can you create a new one? And please describe the system used to record data, and share a file if possible that will be very useful.