sccn/eeglab #256: EEGLab not recognizing AMICA plugin

isim95, 2021-02-15:
### Description

Running EEGLab 2020.0 on mac Catalina. amica1.5 folder is under the plugins path, and the amica15mac file permissions have been changed through terminal. However, when I open EEGLab from MatLab, I do not see AMICA as an option under 'Tools'.

<br>
#### Steps to Reproduce

1. Download eeglab & set up
2. download AMICA, place in EEGLab plugins folder, change amica15mac file permissions with chmod 777 amica15mac on terminal
3. open eeglab
4. load dataset
5. filter with range .5 - 50 hz
6. 

<br>
#### Expected behavior:
amica should be an option under 'Tools'

<br>
#### Actual behavior:
AMICA is not an appearing as an option

<br>
#### 2020.0

|   |   |
|----| ---- | 
| OS version     | [Catalina 10.15.7|
| Matlab version | R2020b|
| EEGLAB version | 2020.0 |

arnodelorme, 2021-02-15:
Amica creates its own menu.

Tools > Run Amica

I have just installed the plugin from the plugin manager and I can see it. I did not have the change the permission to the amica15mac file either. It worked right away.

closed 2021-02-15 by arnodelorme

isim95, 2021-02-15:
This doesn't work for me. I installed from the plugin manager and got the attached error. I've tried installing both ways and am unable to see the Run Amica option under Tools.

[image]

isim95, 2021-02-15:
Apologies, forgot to thank you for the original response! Thank you!

arnodelorme, 2021-02-15:
Would you mind to try downloading it from this page?

https://sccn.ucsd.edu/eeglab/plugin_uploader/plugin_list_all.php

isim95, 2021-02-15:
I got this to work, thank you, and apologies for the inconvenience.

However, I'm not getting the attached error when actually running amica that I'm not quite sure what to do with. Are you potentially able to help with this?
[image]